desowin/usbpcap #18: Icon and wiki?

Syphurith, 2015-02-04:
**This is actually not an issue.**
I'm sorry for sharing something where only issue should goes.
Well most of google services are blocked here so i can't reach your mail list..
I'd got an idea minutes ago and want to show you an icon for this, along with its source illustrator file (which is indeed created by AI CC but converted to a lower version), and yep an icon generated using online service. It is a little ugly - maybe better than nothing. You can get it here: [ZIP](http://pan.baidu.com/s/1hqJ3jfA)
This one would help those hardware developers who are finding a debugger for USB. I see you posted the link to wiki on the page, however there is none article on it (orz). Well only those who have read the source and experienced it a lot can write very good wiki documents..
And.. I'm so busy and i'm quite a noob dealing with hardware and low-level that can't help more. Hope this would go much further than i ever expected.

desowin, 2015-06-23:
Sorry for very late response.

I have just checked your icon and I quite like it. However, I would prefer to now have the USBPcap text on it (it looks really bad for 32x32 icons). Could you please modify it to have a bit bigger main logo and no text (the hex itself is ok)?

Please also include your full name and email address you would like to be attributed with.

Syphurith, 2015-06-24:
[Here](http://pan.baidu.com/s/1hqnKIhe) comes the modification and download password is `swv2`
The .ai file is included, you could do anything you want with it using Illustrator (Well i used CC ver.).
Just modified it a bit, and find my painting skill is bad that i could not get the main part much bigger.
All icons produced using the .png exported from .ai, and then generated using a free online converter.
Newer ones got names such as UsbPcap.NoText.ico, which should include 4 chosen size in the file.
Text is removed successfully, while icon's width be the one of the USB logo that formed from paths.
So unfortunately i could not make it bigger, so sorry.
For name and email .. well i don't deserve the name, just say "one feels it good" would be ok.