desowin/usbpcap #19: USB port(s) not showing up after install, Win7Pro

Progman3K, 2015-02-17:
Hi,

I installed usbpcap on my Windows machine and it DID function until I uninstalled it.

After re-installing/rebooting, the only ports that Wireshark lists as interfaces are the computer's network card and virtual (vitualbox) network card.

What can I do to get it working again?

Is there any log of why it is failing?

Once I install usbpcap it does not function but if I try re-installing it, it complains it must be uninstalled first, so I uninstall it and then reboot, boot up and install usbpcap but it still won't list the USB in the Wireshark interfaces.

Please advise
Thanks

desowin, 2015-06-23:
Does USBPcapCMD list the USB devices?

tallship, 2016-04-29:
I  believe this may be the bug report that has the fix for you here:
https://github.com/desowin/usbpcap/issues/3

desowin, 2017-09-01:
Most likely this is fixed in 1.2.0.3. Please reopen if this is not the case.

closed 2017-09-01 by desowin