desowin/usbpcap #22: How to get full amount of data transferred via USB 3.0 port?

amymlee, 2015-07-19:
Hello. I am using pcap to collect the amount of data that is transferred via USB port through a particular action, such as writing a file into a USB stick. I used pcap to do a live capture of a 1 GB reading into a USB stick, but when I opened the pcap file in Wireshark, the packets are truncated to 65535 bytes and the total number of bytes sums up to less than 1 GB under the "Length" column in Wireshark. Is there a column in wireshark that I can add or a feature in pcap where I can get the data for the actual amount of bytes that are passed through USB instead of truncated packets?

ecbepple, 2015-07-22:
I've run into this as well. I'm trying to capture 160K from a USB eval board and can only see the first 65535 bytes plus a little that shows as "Leftover Capture Data".

thtomate, 2016-06-17:
workaround/solution in #26: https://github.com/desowin/usbpcap/issues/26#issuecomment-223918613

desowin, 2019-05-29:
If you want to capture more than 65535 in single pcap packet, increase the snaplen parameter. It can be configured from Wireshark GUI when USBPcapCMD is copied to Wireshark's extcap directory.

closed 2019-05-29 by desowin