desowin/usbpcap #26: Only 65535 bytes per packet are captured

Videogamer555, 2015-11-08:
So I used the technique shown at http://desowin.org/usbpcap/tour/stdin_capture.png to do live capture (rather than record then analyze). Not sure if it's unique to live capture, or if it's a bug in USBPcap, but if it is a bug, it's pretty major. I was trying to reverse engineer the method that my webcam uses to send image data to the PC, so as to hopefully be able to create my own software that runs with it (rather than having to go through the DirectShow or VFW framework in Windows, or otherwise depend on preexisting drivers). Now this one packet (presumably one containing image data) is 313159 bytes in size as reported by Wireshark, but in another panel in Wireshark it reports "65535 bytes captured", so I can only assume that USBPcap driver reported the size of the USB packet as 313159 bytes, but due to some bug in USBPcap, it failed to capture all of those bytes.

AtlasHackert, 2016-04-11:
It's a feature. See USBPcapDriver/USBPcapMain.h, line 30: `#define USBPCAP_DEFAULT_SNAP_LEN  65535`

However, it does appear there's a way build-in to override it through the command-line: --snaplen

vjerbic, 2016-05-20:
Hello,

Do you know how to change USBPCAP_DEFAULT_SNAP_LEN through command line?
I tried many combinations and still didn't succeed. (-s and --snaplen)

i.e.

_usbpcapcmd --device \\.\USBPcap3  --snaplen 262144 -o  - | "C:\Program Files\Wireshark\Wireshark.exe" -i USBPcap3 -k_
I tried many combinations but Wireshark still captures 65535 bytes. 

Please can you help?

AtlasHackert, 2016-05-20:
It seems the snaplen is limited to: `{range=0,65535}`. I think you'll have to increase the USBPCAP_DEFAULT_SNAP_LEN setting in the source code, and recompile...

vjerbic, 2016-05-20:
Thank you for answer.
The thing is that I don't have win sdk and MS Visual studio installed so i cannot recompile code 
The snaplen I need is 512kB.

vjerbic, 2016-05-23:
I installed Visual Studio Community 2013 and Windows Driver Kit 8.1 in order to redcompile code with different snaplen.

I am running on Windows 8.1 Proffesional 64-bit.

I tried to recompile using driver_build_win8.bat but it failed with following errors.
[image]

Is it because I am running on Win 8.1 or something else? 
Please help how to recompile code?

AtlasHackert, 2016-05-23:
That's a header file that should be included from the Windows SDK. I suspect you either didn't install all the needed components from the SDK, or it is set-up incorrectly in your Visual Studio.

Videogamer555, 2016-06-05:
Can you do me a favor? Since I REALLY want to use this, to intercept
communicatoins with my webcam, I will need to be able to fit an entire
frame of YUV420 image data at 640x480 image size, into the size that USBCAP
is capable of capturing. Needless to say, 65535 is WAY to small for this.
Preferably the maximum packet size should set by the range= command, with
NO UPPER LIMIT as to how high you can set the maximum packet size with that
command. So the favor I am asking of you, is to find a person who's
proficient in programming with C++, contact them with my request, and see
if they will be willing to compile a custom version of this plugin for me,
and then if it's small enough send the compiled copy to me it via email
attachment, or if it's too big then upload it to Mediafire, and me the
download link in an email.

AtlasHackert, 2016-06-05:
I'm currently not set-up to compile it, but I see somebody else who might be:
https://github.com/mmann78/usbpcap/commits/master
Maybe you can ask him/her?

vjerbic, 2016-06-06:
I could not solve problem to recompile code.
However, I realized that only USBPcap that came with Wireshark cannot capture with different snaplen.

I unistalled that version and installed version from http://desowin.org/usbpcap/ site.
Using command line, I succesfully captured packets with larger packets and store them to file with pcap extension.
But when I tried to open them with Wireshark, error apperead.
It looks like Wireshark cannot open file which has packets larger than 65535 bytes.

I used Matlab to extract neccessary data from pcap file.

commit 2cb46e38 (2019-05-29): Remove extcap snaplen argument range

Change the type to unsigned to allow full unsigned integer range.

This makes it possible to set the snaplen to whatever value the user
wishes directly from the Wireshark GUI.

This closes #26
closed 2019-05-29 by desowin