desowin/usbpcap #27: Installation issue - System Restore

ptoye, 2015-12-01:
I installed USBPcap as part of Wireshark. During the installation I got a window saying that System Restore had restored my system to a previous restore point. Rather worrying, the date on the restore point was earlier than any of the dates on restore points shown by System Restore. 

I'm not sure yet whether any data was lost, but even if it wasn't, this isn't a good idea.

RUnning Windows 7 Home Premium SP1.

desowin, 2019-05-29:
I don't understand what you are reporting. Issue does not contain any version information so it doesn't help. If you can reproduce the issue with recent USBPcap installer (1.4.0.0 or newer), please reopen.

closed 2019-05-29 by desowin

ptoye, 2019-05-29:
Marvellous. 3 years after I made this comment someone decides to close it down. If you don't understand it, wouldn't it be more polite to ask me what I meant rather than taking a unilateral decision?