desowin/usbpcap #31: USBPcap/KB3033929

Jean-MarcL, 2016-05-03:
Hi, I installed the last version of WireShark (2.0.3) with USBPcap embeded. During the installation it requires the KB3033929 package. After that I Installed the KB3033929 and tried to install again USBPcapSetup-1.1.0.0-g794bf26-3.exe. But it gives the same message, KB3033929 required!!! KB3033929 is really installed, I tested it. Is there a way to install it manually? ( I can extract the files with 7zip...) Thanks!!!

thtomate, 2016-07-11:
Same here.
Hotfix was already installed.
I already used it on this system some time ago, but had to deinstall it, because of the conflict with VirtualBox.

skleeschulte, 2016-09-14:
I have a similar problem: When I try to install Wireshark with USBPcap option enabled or to install C:\Program Files\Wireshark\USBPcapSetup-1.1.0.0-g794bf26-5.exe later, I always get the message `Hotfix KB3033929 must be installed on Windows 7 or 2008R2.`.

I downloaded Windows6.1-KB3033929-x64.msu and tried to install the update, but this gives me the message `Security Update for Windows (KB3033929) is already installed on this computer.` I tried it several times with rebooting and stopping / starting windows update and even reset windows update with the [Reset Windows Update Agent](https://gallery.technet.microsoft.com/scriptcenter/Reset-Windows-Update-Agent-d824badc) tool.

I cannot find KB3033929 in the list of installed Windows updates. I found [a question "USBPcap/KB3033929" on wireshark.org](https://ask.wireshark.org/questions/52166/usbpcapkb3033929) where it is mentioned that the USBPcap installer uses the command `%COMSPEC% /C wmic qfe get Hotfixid | findstr KB3033929` to check if KB3033929 is installed. If I run this command in an elevated command prompt I get an empty result (indicating a missing KB3033929 hotfix).

I guess the problem might be that KB3033929 is superseded by another windows update and thus was never installed and is still reported as already being installed when trying to install the update. I found the information that this scenario is possible in [Microsoft Security Advisory 3033929](https://technet.microsoft.com/en-us/library/security/3033929?f=255&MSPPError=-2147217396), although here it is explained the other way round - KB3033929 superseding another update:

> Scenario 2: Customer first installs advisory update 3033929 and then attempts to install update 3035131.
> Result: The installer notifies the user that the 3035131 update is already installed on the system; and the 3035131 update is NOT added to the list of installed updates.

I recently installed my Win 7 machine using Win 7 SP1 installation media and immediately afterwards installing the [Convenience rollup update for Windows 7 SP1 and Windows Server 2008 R2 SP1](https://support.microsoft.com/en-us/kb/3125574). Unfortunately I don't know how to find out which windows update might supersede KB3033929.

Any ideas on this with this new information?

amigabill, 2016-10-20:
Any update? I have same problem. usbpcap won't install, and I cannot install the patch as it is "already installed".

amigabill, 2016-10-20:
What would one do for a fully manual install in Win7 64bit, to get around this problem?

Jean-MarcL, 2016-10-24:
USBPcap is now provided with the last versions of WireShark.
The current version is USBPcapSetup-1.1.0.0-g794bf26-5.exe in the WireShark Setup. 
As far as I remenber, I never succeded with the installation on http://desowin.org/usbpcap/

skleeschulte, 2016-10-26:
For me, the USBPcap installer that ships with WireShark doesn't work either (see above).

amigabill, 2016-10-27:
I installed it from current wireshark downloaded a couple days ago. google
got me a thread in wireshark forums that said to report the problem tobus
cap forum onstead, so i did...

pengc99, 2017-01-24:
I ran into the same problem. Here's the workaround:

1. Download Wireshark executable and install. Make sure to select usbpcap, it will fail.
2. Download 7zip executable and install (http://www.7-zip.org/download.html)
3. Download usbpcap executable and install (http://desowin.org/usbpcap/)
3. Use the right-click context menu to extract Wireshark installer
4. Browse into Wireshark installer extracted folder
5. Extract USBPcapSetup-1.1.0.0-g794bf26-5.exe from the root of Wireshark installer folder
6. Browse into the USBPcapSetup-1.1.0.0-g794bf26-5 folder
7. Copy USBPcapCMD.exe 
8. Paste executable into C:\Program Files\Wireshark\extcap (replace existing copy of executable, create folder if it doesn't exit)
9. Reboot computer
10. Start Wireshark as administrator

This loads the signed driver in the latest stand-alone usbpcap release and uses the capture plugin from the Wireshark release.

CaptainFlint, 2017-04-17:
Same problem here. I used a nice hack to fool the installer. Effectively, I created a substitute findstr that reported that KB3033929 line is found in the output.
1. Create a file named findstr.cmd with one line:
`@echo KB3033929`
2. Copy this file to c:\windows\syswow64 on 32-bit system or to c:\windows\system32 on 32-bit system.
3. Rename c:\windows\syswow64\findstr.exe (or system32, respectively) into something line findstr1.exe. You'll need to change the file owner to yourself and add full permissions to yourself for this file.
4. Run the USBPcap installer - now it should run successfully.
5. Delete findstr.cmd, rename findstr1.exe back and restore the permissions. In my case its original owner was TrustedInstaller which you must enter as NT SERVICE\TrustedInstaller, otherwise the name will njot be accepted.

Tested on Win7 x64.

Green7, 2017-04-29:
I have this same problem with my app.
Wmic is unreliable  so instalator should check wintrust.dll version.
If the version is  >= 6.1.7601.18741, there is no need to install KB3033929.
Here is similar issue and solution: https://github.com/dokan-dev/dokany/pull/328

commit 4746a708 (2017-05-04): Check for GDR/LDR version of KB3033929

Fixes #31
closed 2017-05-04 by desowin