desowin/usbpcap #33: desowin.org is down

fstd, 2016-11-22:
is there an alternative download mirror?

desowin, 2016-11-22:
https://bintray.com/desowin/USBPcap
https://sourceforge.net/projects/usbpcap/?source=directory

Installer is also bundled with Wireshark.

fstd, 2016-11-22:
On Tue, Nov 22, 2016 at 06:39:32AM -0800, Tomasz Moń wrote:
> https://bintray.com/desowin/USBPcap
> https://sourceforge.net/projects/usbpcap/?source=directory
Thanks!

> Installer is also bundled with Wireshark.
That one bails out telling me I need KB3033929, when the
KB3033929 installer tells me it's already present, duh :-).
That's how I came to be looking for an alternative source
in the first place.
This is Win7 Enterprise 64bit, in case it matters.

desowin, 2016-11-23:
Added all releases to Github Releases. Website is now hosted on Github Pages.

closed 2016-11-23 by desowin