desowin/usbpcap #34: Does not see data to/from MIDI device

snhirsch, 2016-12-12:
I'm trying to watch communication between a Boss Katana 100 guitar amplifier and the "Boss Tone Studio" software used to edit operating parameters on the device.  I can see the amplifier listed on hub #2, but USBPCap does not capture anything.  I get only a 24-byte file and a report that no data was seen.  Have tried this on 64-bit Windows 10 and 64-bit Windows 7.  No luck on either.

A commercial program called USBLyzer is able to do the job, but it's both time and feature limited.  Any ideas how I can work around this problem?

desowin, 2019-05-26:
Can you try with USBPcap 1.4.0.0? Does the capture contain any unknown transfer type (0xFF)?

snhirsch, 2019-05-26:
Well, it's about 2-1/2 years later but I do thank you for the response!  I will try to set this all up again in the near future and see if a newer USBPcap can get any further.

ArielVinda, 2020-04-26:
Got latest version, and tried with my katana mkII, same deal as above