desowin/usbpcap #37: How to Redirect USBPcapCmd.exe "USB interfaces list" output to a text file.

pramodjnp, 2017-05-31:
Hi i need to redirect interface list from USBPcapcmd.exe to an text file for getting USB HUB number in command line. please help me how to achieve this, is there any way, i can get "USB interface list" to a text file.

please any one help me

Thanks in Advance!!

desowin, 2017-06-01:
If you would like a easily parsable string then please take a look at extcap interface specification: https://wiki.wireshark.org/Development/Extcap#Current_flow

USBPcapCMD implements extcap interface. Examples:
USBPcapCMD.exe --extcap-interfaces
USBPcapCMD.exe --extcap-interface \\.\USBPcap1 --extcap-config

pramodjnp, 2017-06-01:
Hi thanks for the reply. Is there any way we can get **"interface"** with its connected **"USB Drives list"**.
As there in the below image.

[image]

Thanks in Advance!!!

pramodjnp, 2017-06-05:
Hi desowin, 
Thanks for the reply, its working for me.

closed 2017-06-05 by pramodjnp