desowin/usbpcap #38: Installer pops-up a malicious file warning on Windows 10

gbloice, 2017-05-31:
Running Windows 10 Creators Edition 107, build 15063, attempting to run the 1.2.01 installer causes a large red "Malicious file" pop-up to appear.

I wonder if this is because the installer is only signed with a SHA-1 certificate with a SHA-1 digest hash?

desowin, 2017-06-01:
I think it has something to do with the fact that the certificate is brand new and has not been used to sign any files before (USBPcap 1.2.0.1 is the first installer/driver signed with that certificate). Also, it is Standard Code Signing, not the EV Code Signing. Unfortunately EV Code Signing certificate cannot be issued to individuals.

From what I have read on the internet about Windows SmartScreen it is a matter of time until installer builds enough trust (based on number of people actually installing and using it). It should be easier with future releases (as the certificate itself hopefully will get some trust) as long as they will be signed with the same certificate.

gbloice, 2017-06-01:
I hadn't seen such a notification before so it surprised me.  I've not even seen (or had reported) such a notice for our own signed code (also with a standard cert) even on the day the cert was issued.  We have dual-signed (SHA-1 & SHA-256) ever since Windows 10 though.

I don't see the issue on an older Win 10 system (1511) that does have SmartScreen enabled, so either the problem has gone away or it's only evident on Win 10 1703.

While an EV cert *might* help, we don't use our EV cert on our installers or even on drivers, you just need to upload the EV cert to the Windows Dev Center site to prove you have one, and then sign the drivers with the standard cert before uploading them to MS for their signature.

desowin, 2018-08-14:
USBPcap 1.2.0.4 is dual signed using standard code signing certificate. This might help, but only time will tell.

closed 2018-08-14 by desowin