desowin/usbpcap #40: USBPcap bug faults with driver verifier enabled

desowin, 2017-07-04:
Originally reported by @JimNickerson in #3. I can reproduce it on Windows 10.

0x23E
Non-fatal error
A driver has marked an IRP pending but didn't return STATUS_PENDING. (IRP specified.)

Workaround:
Disable driver verifier

closed 2017-08-01 by desowin

desowin, 2017-08-01:
@JimNickerson: Could you please confirm that 1.2.0.2 fixes the issue for you?

JimNickerson, 2017-08-01:
I downloaded 1.2.0.2
I was presented with this ominous box when I tried to run it.
Suggesting it was Malicious
[image]

desowin, 2017-08-01:
Unfortunately this is known issue. It will get waived once few people install it. This unfortunately is how it works with my certificate.

JimNickerson, 2017-08-01:
Ok, I removed the old wireshark and installed the current with USBPcap 1.2.0.2
When I start Wireshark it hangs on "Initializing extcap"
I can start USBPcapCMD.exe manually.
I get
Following filter control devices are available:
1 \\.\USBPcap1
  \??\USB#ROOT_HUB20#4&27638150&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}
    [Port 1] Generic USB Hub
2 \\.\USBPcap2
  \??\USB#ROOT_HUB20#4&2c6ffe96&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}
    [Port 1] Generic USB Hub
3 \\.\USBPcap3
Couldn't open device - 5
Select filter to monitor (q to quit):
If I leave USBPcapCMD.exe window open I can then start Wireshark, but there is no traffic displayed.
If I double click on Port 1, Wireshark responds with Unrecognized libpcap format or not libpcap data
And the next dialog says Error by extcap pipe C:\Program Files\Wireshark\extcap\USBPcapCMD.exe:--devices option requires an option argument.

JimNickerson, 2017-08-01:
If I rename Program Files\USBPcapCMD.exe to .exez
And terminate the process left running
I then can start Wireshark normally

JimNickerson, 2017-08-01:
If I start USBPcapCMD manually, specify a port, wait for a bit, ^C, double click on the .pcap file wireshark opens it and displays it
I now get more information on Port 3, How do I select the downstream ports ?
Following filter control devices are available:
1 \\.\USBPcap1
  \??\USB#ROOT_HUB20#4&27638150&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}
    [Port 1] Generic USB Hub
2 \\.\USBPcap2
  \??\USB#ROOT_HUB20#4&2c6ffe96&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}
    [Port 1] Generic USB Hub
3 \\.\USBPcap3
  \??\USB#ROOT_HUB30#4&e100702&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}
    [Port 3] Generic USB Hub
      [Port 1] Generic USB Hub
      [Port 3] Generic USB Hub
    [Port 5] Logitech USB Camera (Pro 9000)
      Logitech QuickCam Pro 9000
      Pro 9000
        Microphone (Pro 9000)
    [Port 6] Generic USB Hub
      [Port 1] USB Input Device
        HID-compliant mouse
      [Port 3] USB Input Device
        Kensington Expert Mouse
          TrackballWorks 01
    [Port 7] Realtek USB 2.0 Card Reader
    [Port 8] USB Input Device
      HID-compliant vendor-defined device
    [Port 11] Intel(R) Wireless Bluetooth(R)
      Microsoft Bluetooth LE Enumerator
      Bluetooth Device (RFCOMM Protocol TDI)
      Microsoft Bluetooth Enumerator
        PLT_Legend Stereo
        AV Remote Target Service
        PLT_Legend Audio/Video Remote Control HID
        PLT_Legend Hands-Free Audio and Call Control HID Enumerator
          PLT_Legend Hands-Free
          PLT_Legend Hands-Free Call Control HID
        PltHeadsetDataService
        PLT_Legend
      Bluetooth PAN HelpText
    [Port 13] USB Composite Device
      USB Input Device
        HID Keyboard Device
      USB Input Device
        HID-compliant system controller
        HID-compliant consumer control device
        HID-compliant device
    [Port 14] Generic USB Hub
    [Port 18] Generic SuperSpeed USB Hub
      [Port 1] Generic SuperSpeed USB Hub
      [Port 3] Generic SuperSpeed USB Hub
        [Port 3] USB Mass Storage Device
          Generic STORAGE DEVICE USB Device
          Generic STORAGE DEVICE USB Device
          Generic STORAGE DEVICE USB Device
Select filter to monitor (q to quit):

commit 0f60886f (2017-08-01): Propagate IRP status in DkTgtInDevCtlCompletion()

This seems to solve BSoD on Windows 10 with Driver Verifier enabled.
Fixes: #40