desowin/usbpcap #42: Malformed packets that Windows interprets as okay

kuro68k, 2017-07-17:
I am trying to debug an issue with a device I built. When I connect it, Windows can read the descriptors and strings okay. It shows up in Device Manager as "Example Device" (the product ID I set) and the properties page shows the correct VID/PID, Manufacturer ID etc. Other USB debug tools show that the descriptor was read correctly (vendor specific device with bulk endpoint).

However, in Wireshark the usbpcap capture shows "malformed packet" on all the descriptors and strings.

```
Frame 604: 30 bytes on wire (240 bits), 30 bytes captured (240 bits)
    Encapsulation type: USB packets with USBPcap header (152)
    Arrival Time: Jul 17, 2017 10:05:00.699619000 GMT Daylight Time
    [Time shift for this packet: 0.000000000 seconds]
    Epoch Time: 1500282300.699619000 seconds
    [Time delta from previous captured frame: 0.000000000 seconds]
    [Time delta from previous displayed frame: 0.000000000 seconds]
    [Time since reference or first frame: 14.331000000 seconds]
    Frame Number: 604
    Frame Length: 30 bytes (240 bits)
    Capture Length: 30 bytes (240 bits)
    [Frame is marked: False]
    [Frame is ignored: False]
    [Protocols in frame: usb]
USB URB
    [Source: 1.2.0]
    [Destination: host]
    USBPcap pseudoheader length: 28
    IRP ID: 0xfffffa800a9af5b0
    IRP USBD_STATUS: USBD_STATUS_SUCCESS (0x00000000)
    URB Function: URB_FUNCTION_CONTROL_TRANSFER (0x0008)
    IRP information: 0x01, Direction: PDO -> FDO
        0000 000. = Reserved: 0x00
        .... ...1 = Direction: PDO -> FDO (0x1)
    URB bus id: 1
    Device address: 2
    Endpoint: 0x80, Direction: IN
        1... .... = Direction: IN (1)
        .000 0000 = Endpoint value: 0
    URB transfer type: URB_CONTROL (0x02)
    Packet Data Length: 2
    [Request in: 603]
    [Time from request: 0.000000000 seconds]
    Control transfer stage: Data (1)
STRING DESCRIPTOR
    bLength: 70
    bDescriptorType: 0x03 (STRING)
[Malformed Packet: USB]
    [Expert Info (Error/Malformed): Malformed Packet (Exception occurred)]
        [Malformed Packet (Exception occurred)]
        [Severity level: Error]
        [Group: Malformed]
```

```
0000   1c 00 b0 f5 9a 0a 80 fa ff ff 00 00 00 00 08 00  ................
0010   01 01 00 02 00 80 02 02 00 00 00 01 46 03        ............F.
```

The request from the host is for string 0x02, which is the product ID, which in this case should be "Example Device", which is what Windows displays.

kuro68k, 2017-07-17:
You can download the capture here: https://drive.google.com/file/d/0Bxo3omeSKZ7AcWVTdUVacFpWOWM/view?usp=sharing

kuro68k, 2017-07-17:
I also checked with a logic analyzer, I can see the correct packets and data being sent.

desowin, 2017-07-17:
Which Windows version are you using?

kuro68k, 2017-07-17:
Windows 7 x64 Pro.

kuro68k, 2017-07-17:
I can confirm that Microsoft Network Monitor works and captures the packets correctly.

desowin, 2019-04-01:
I believe this was fixed in fbef41944d369680765bb7d82d2cc8c3b9455d32

The USBPcap 1.3.0.0 release should no longer miscapture these packets.

closed 2019-04-01 by desowin

inferiorhumanorgans, 2021-03-21:
I'm running USBPcap 1.5.4.0 (Win 7 SP1/x86 under VirtualBox) and Wireshark v3.4.3-0-g6ae6cd335aa9 (OSX 10.14.6) and am seeing:

```Frame 725: 28 bytes on wire (224 bits), 28 bytes captured (224 bits)
USB URB
    [Source: 1.2.1]
    [Destination: host]
    USBPcap pseudoheader length: 27
    IRP ID: 0xffffffff854e29a8
    IRP USBD_STATUS: USBD_STATUS_SUCCESS (0x00000000)
    URB Function: URB_FUNCTION_BULK_OR_INTERRUPT_TRANSFER (0x0009)
    IRP information: 0x01, Direction: PDO -> FDO
    URB bus id: 1
    Device address: 2
    Endpoint: 0x81, Direction: IN
    URB transfer type: URB_BULK (0x03)
    Packet Data Length: 1
    [Request in: 722]
    [Time from request: 0.000000000 seconds]
    [bInterfaceClass: Vendor Specific (0xff)]
[Malformed Packet: IPPUSB]
```

Netmon is decoding the packet and showing a one byte payload.  Wireshark is showing "Packet Data Length" as "01 00 00 00" with a trailing byte that's being classed as a malformed packet.  The device itself appears to be functioning as the bundled app communicates with it appropriately.

desowin, 2021-03-21:
> I'm running USBPcap 1.5.4.0 (Win 7 SP1/x86 under VirtualBox) and Wireshark v3.4.3-0-g6ae6cd335aa9 (OSX 10.14.6) and am seeing:
> 
> ```
> USB URB
>     [Source: 1.2.1]
>     [Destination: host]
>     USBPcap pseudoheader length: 27
>     IRP ID: 0xffffffff854e29a8
>     IRP USBD_STATUS: USBD_STATUS_SUCCESS (0x00000000)
>     URB Function: URB_FUNCTION_BULK_OR_INTERRUPT_TRANSFER (0x0009)
>     IRP information: 0x01, Direction: PDO -> FDO
>     URB bus id: 1
>     Device address: 2
>     Endpoint: 0x81, Direction: IN
>     URB transfer type: URB_BULK (0x03)
>     Packet Data Length: 1
>     [Request in: 722]
>     [Time from request: 0.000000000 seconds]
>     [bInterfaceClass: Vendor Specific (0xff)]
> [Malformed Packet: IPPUSB]
> ```

This looks like a bug in Wireshark, but I would need complete pcap file to check what's exactly going on.

inferiorhumanorgans, 2021-03-21:
I'll email the pcap and ETL files in a moment.