desowin/usbpcap #43: Windows 10 flagging unsigned driver

herrameise, 2017-07-20:
The installer may be signed, but it looks like the actual driver is not. After installing USBPcap 1.2.0.1, Windows 10 displays a message that it has blocked the installation of an unsigned driver. On reboot, all of my USB devices no longer worked until I uninstalled the USBPcap program. This was not an issue with the older version of USBPcap that used to ship with Wireshark. However, the new Wireshark release (2.4.0) is bundled with USBPCap 1.2.0.1.

herrameise, 2017-07-20:
just verified that the previous release (1.0.0.7) has no issues installing on Windows 10

desowin, 2017-07-20:
I am pretty sure the driver is signed. It is standard Code Signing certificate however.

I am working on HLK submission which should solve the problem.

jeremypoulter, 2017-08-01:
I also had this issue, is there any update?

desowin, 2017-08-01:
Please report how USBPcap 1.2.0.2 behaves on your computer.

jeremypoulter, 2017-08-03:
Sorry, I get the same;

[image]

desowin, 2017-08-04:
Do you have Secure boot enabled? Did the USB devices work after restart?

The solution to this is attestation signing, but unfortunately I cannot do this alone, it needs to go through the company and there is paperwork burden.

desowin, 2017-08-21:
Please let me know if 1.2.0.3 works for you.

rogerdc, 2017-08-22:
I was able to successfully install it and my USB devices are still working.

closed 2017-09-01 by desowin