desowin/usbpcap #44: Erroneous "Hotfix 3033929 must be installed on Windows 7 or 2008R2"

cde-, 2017-09-29:
The installer tries to detect this required march 2015 hotfix. However, on this installation of Windows 7 SP1 the 2016 "Convenience rollup update for Windows 7 SP1" is installed and does contains the hotfix (as evidenced by the KB3033929 hotfix saying it is already installed when trying to install it).

Solution: either remove the call to exit in the installer when the hotfix is not found, or accept installation when the convenience rollup update is installed (KB3125574).

desowin, 2017-09-29:
Which version are you using?

This shouldn't happen in USBPcap 1.2.0.3

cde-, 2017-09-29:
Thanks, I was indeed using an older version, USBPcapSetup-1.1.0.0-g794bf26-5.exe (bundled with Wireshark 2.2.5). I can confirm the latest Wireshark contains an updates USBPcap that does not show this error message.

closed 2017-09-29 by cde-