desowin/usbpcap #47: Windows 10 Installer Thinks This App is Malware

MelbourneDeveloper, 2017-12-26:
There is no reason to suspect that this app is malware especially considering that it is open source, and the installer is digitally signed. Yet, Windows 10 gives you an ugly red Malware warning when you try to install. The certificate is issued by Certum. The signer is Tomasz. Is there any way to sort out the certificate problem to satiate Windows 10's desire for certainty?

This screen gives me the heebie jeebies:
https://www.dropbox.com/s/cs1jpri7jjflyc9/Malware.png?dl=0

desowin, 2018-08-07:
Yes, USBPcap since release 1.2.0.3 contains Attestation Signed driver which works on Windows 10.

The way to "sort out the certificate problem to satiate Windows 10's desire for certainty" is to get a fiscal sponsor which would take donations for a project and then obtain EV Code Signing certificate. This certificate is immediately trusted by SmartScreen. Unfortunately it is unavailable for individual developers. Registering an organization in my country just to get through this is way too much legal burden for me. Maybe there are countries where registering the organization wouldn't take as much time/money, so a single individual can manage it in his free time (solely to obtain the certificate and sing official releases from time to time).

I tried applying to Software For Public Interest and Software Freedom Conservancy. Unfortunately didn't succeed. However, there's a possibility that these organizations might be interested in the future - the project would need to change from "single hero" to "multiple contributors". This is something where I need contributions from others.

The list of places where contributions are most needed:
- Website. especially the tour. It'd be very nice to include the description about Wireshark extcap integration. Also some common questions/problems would be nice - some information about it can be obtained by looking through the comments on github issues.
- Creating .dll/static-library to be used by USBPcapCMD *and* which would provide an easy interface for external projects.
- Fixing some bugs in the driver. The control transfers DATA phase is not captured if it is going from host to device. Also on Windows 8, there are bulk and isochronous transfers which use chained MDLs which are not captured at all. The bugs in the driver are my next goal after I release 1.2.0.4.

MelbourneDeveloper, 2018-08-12:
@desowin thankyou very much for the detailed reply. 

This really does pose a major problem for your software and other pieces of software that may want to do the same thing. Is there a reason that normal code signing won't do the trick? Is it because your software calls APIs that are deemed too volatile by the operating system?

desowin, 2018-08-13:
I don't know how really this "gaining SmartScreen reputation" works. The 1.2.0.4 will be released next week, signed both with SHA1 and SHA256 (Certum fixed the bug in their ProCertum Software which made it possible only to have one cert on the card at the same time - which was the reason for previous release being SHA1 only). So if that's going to help, only time will tell.

It doesn't matter what API you call.

You need kernel mode certificate for the driver and for Windows 10 driver you need to do Attestation Signing (possible only with EV Code Signig). Hopefully my employer allows me to do the attestation sining of USBPcap using company certificate, so it's not as hopeless as it might look like.

But sure, the hassle related to certificates (they are quite expensive) is in my opinion the biggest demotivation factor for individuals like me doing Windows OSS software.

desowin, 2018-08-14:
I have absolutely no idea why Windows 10 marks USBPcap 1.2.0.4 as Trojan:Win32/Spursint.F!cl. The only possibility would be that I installed infected NSIS or that it is a false positive.

https://www.virustotal.com/#/file/0a5ac30b0264e058f262e9c28e5865af7b836620ca5d68bb4bb42c9a808f7a43/detection

I have submitted the file to Avast whitelisting, I hope there isn't any malware that slipped onto my build machine.

desowin, 2018-08-14:
I have submitted the installer to Windows Defender Security Intelligence.

desowin, 2018-08-14:
Submitted the installer to Antiy-AVL false positive reporting email.

desowin, 2018-08-14:
I have received reply from Windows Defender Security Intelligence that they have reviewed the file and removed the detection.

MelbourneDeveloper, 2018-08-14:
Fantastic!

MelbourneDeveloper, 2018-08-14:
@desowin sorry for not being more vocal but I'm stoked about this. My library is https://github.com/MelbourneDeveloper/Hid.Net . I wanted to use your library to reverse engineer USB transfer and this is going to be great. Thanks for doing this. 

I have a bunch of other hardwarewallet libraries based on hid.net so I will be able figure out a lot more once I can trace the transfer.

Thanks!

desowin, 2018-08-15:
What's preventing you from using it? The red SmartScreen warning is supposed to go away once "enough" people download the installer and click the "run anyway" option. What some websites suggest is that every download that doesn't click the "run anyway" is essentially setting it further away from the goal of "gaining reputation".

The fact that they removed the detection refers to the fact that Windows defender simply removed the file as soon as it was downloaded. That was really harsh. Now it's back to just the red warning.

The red SmartScreen is just unfair, you can google a lot about smaller software publishers getting hit by it. The only solution I found online is to tell users to click the "run anyway" button...

desowin, 2018-08-15:
https://www.rizonesoft.com/inconvenient-truth-about-smartscreen/

desowin, 2018-08-20:
There is option to report the download as safe in the Internet Explorer. The method is described in https://www.ctrl.blog/entry/how-to-false-smartscreen-positive

If you would like to help, please download USBPcap installer from the [official website](https://desowin.org/usbpcap/), and then right click on the downloaded file (in IE downloads list) and select "Report that this download is safe". I have checked that this option is available in Windows 7, 8.1 and 10. I hope if enough people do this, SmartScreen will finally stop this false reporting.

desowin, 2018-08-23:
Symantec Endpoint Protection does similar "reputation" checking as SmartScreen, but unlike SmartScreen it doesn't claim that it's malware. Thus such kind of check can be done in a not so completely unfair way as SmartScreen.

[image]

desowin, 2018-09-14:
All the false positives have now been resolved, and the 1.2.0.4 installer is no longer triggered by any scanner on [virtustotal](https://www.virustotal.com/fi/file/0a5ac30b0264e058f262e9c28e5865af7b836620ca5d68bb4bb42c9a808f7a43/analysis/1536914977/)

MelbourneDeveloper, 2018-12-08:
Fixed!!!!

closed 2018-12-08 by MelbourneDeveloper