desowin/usbpcap #51: Wireshark integration: support new extcap option "--extcap-version=x.x"

Wootness, 2018-07-13:
The latest build version of Wireshark (2.9.0) contains a change to the extcap API which breaks compatabiliy with usbpcap.

Until now, USBPcapCMD.exe was queried for it's extcap interfaces using:
"USBPcapCMD.exe --extcap-interfaces"
But from version 2.9 it will change to:
"USBPcapCMD.exe --extcap-interfaces --extcap-version=2.9"

Currently running USBPcapCmd.exe with these args returns this error:
"USBPcapCMD.exe: --extcap-version: unknown option"
This leads to wireshark rejecting the executable and not allowing capturing using it's GUI.

desowin, 2018-07-25:
Before there is a new release, one might want to use the latest AppVeyor build of USBPcapCMD (x64 only; Windows 10 warns about lack of signature - but works).

Wootness, 2018-07-25:
For what it's worth, this also works on my (Win10 x64) machine with the master branch of Wireshark 2.9

desowin, 2018-07-27:
Conversation moved from [Pull Request](https://github.com/desowin/usbpcap/pull/53)

@guyharris wrote:
> Note that 1) the version argument to --extcap-version is optional, so if there's an argument to the --extcap-version option, it shouldn't print the extcap version (it should presumably check whether it supports that version and fail if it doesn't), and if there isn't an argument to that option, it should continue to report the version.

>Note also that the only way to pass an optional argument to an option that is supported by all versions of getopt_long() is --extcap-version=X.Y, not --extcap-version X.Y, so Wireshark will pass it as --extcap-version=X.Y; if you're not using getopt_long() or something compatible with it as an option parser, you must also support --extcap-version=X.Y.

To which I have replied:

I will change the argument parser to support the optional argument.

However, I am somehow lost in understanding --extcap-version. Could you please answer following questions to clarify:

When Wireshark calls --extcap-version=X.Y then the X.Y is the Wireshark version, eg. 2.9, right? Is this "should presumably check whether it supports that version and fail if it doesn't" supposed to check if the wireshark version is too old? For example, assume that some extcap makes it mandatory to use some extcap feature that is supported in Wireshark since some version A.B. Then the extcap will check the passed X.Y against the A.B and if it is lower, it'll fail. Is this the intention?
If wireshark calls --extcap-version without any argument, the extcap should print its version (eg. 1.2.0.4 for the upcoming USBPcap), right?
Regardless if --extcap-version is called with or without parameter, if extcap does not fail, it should print the tool version (the 1.2.0.4 for upcoming USBPcap)?

And if extcap is called without --extcap-version at all, but with the --extcap-interfaces (old version of Wireshark), is it safe to print the "extcap {version=a.b.c.d}{help=someurl}" alongside the interfaces?

pquantin, 2018-07-27:
My understanding is that --extcap-version parameter is used to indicate the Wireshark version launching the extcap tool (in case the extcap tool has a dynamic behavior depending on the Wireshark version used because of new or changed features for example).
To provide its own version, the extcap tool should return the version=x parameter when called with --extcap-interfaces. As far as I know --extcap-interfaces should not be called without any argument by Wireshark (that should be an error I guess). See https://www.wireshark.org/docs/wsdg_html_chunked/ChCaptureExtcap.html for details.
The version parameter (to provide the extcap tool version) was introduced before --extcap-version so it should be safe to send it when invoked without --extcap-version.

guyharris, 2018-07-27:
Prior to Wireshark 2.9/3.0, the `--extcap-version` option took no argument, and indicated that the extcap program should report *its* version, e.g.

    $ build/run/Wireshark.app/Contents/MacOS/extcap/randpktdump --extcap-version
    extcap {version=0.1.0}{help=file:///usr/local/share/wireshark/randpktdump.html}
    $

Starting in 2.9/3.0, it can *optionally* take an argument.  If no argument is specified, it still means "report the version of the extcap program"; if an argument is specified, it means "this is the version of the Wireshark/TShark/etc. that's running the extcap program".

I think it was perhaps unwise to repurpose an existing option for a separate purpose, so I may just change it so that `--extcap-wireshark-version` is an option with a *required* argument (so that it can either be specified as `--extcap-wireshark-version X.Y` or `--extcap-wireshark-version=X.Y` that indicates what version of *Wireshark* is running the extcap program and `--extcap-version` *never* takes an argument and *always* asks the extcap program to report *its* version.

desowin, 2018-07-28:
The multicheck doesn't work properly in Qt interface. I have checked this in Wireshark 2.0.16 - the GTK+ interface works as expected, however the Qt doesn't even allow selecting the devices. The multicheck is to be fixed in Wireshark Qt interface itself.

desowin, 2018-07-28:
Well, the Wireshark Qt interface doesn't seem to be broken. In fact it does the "select by highlighting" which is confusing to me (press CTRL and click to select multiple ones). I would prefer if it displayed the tickboxes like the GTK+ interface did.

What I have noticed is that it was me who did the [change](https://code.wireshark.org/review/#/c/4192/) that made USBPcapCMD working with GTK+ interface in a way I expected. I didn't realize however, that the implementation didn't fully follow my comment - and I screwed up hard by making all devices in USBPcapCMD --extcap-config to be listed with {enabled=false} (which means they should not be clickable). There's no point in really figuring out what exactly is broken in that change (as GTK+ interface is long gone) and/or trying to fix it.

The problem I listed in previous comment is really related to a corner case where extcap does give multicheck, but then all the options are not selectable.

desowin, 2018-08-14:
USBPcap 1.2.0.4 is now officially relased.

closed 2018-08-14 by desowin