desowin/usbpcap #55: Preparing for USBPcap 1.2.0.4 release

desowin, 2018-07-31:
The USBPcap 1.2.0.4 will feature the same driver as in previous release and modified USBPcapCMD.

There are numerous improvements to USBPcapCMD done already in master branch:

- Fixed full CPU core utilization during capture when using USBPcapCMD as extcap (#32)
- Extra sanity checks during enumeration in order to prevent freezing Wireshark when used as extcap (#50)
- Reworked command line arguments handling so it works with Wireshark 2.9 (#51)
- Fixed USBPcapCMD on Windows XP (it wasn't working at all)
- Fix synchronization problems on exit, now pressing 'q' in the USBPcapCMD console window should exit pretty much immadietely

I would appreciate some pre-release testing. If you are interested, please:

- Either build USBPcapCMD yourself or download it from [AppVeyor](https://ci.appveyor.com/project/desowin/usbpcap)
- Check if it works on your computer
- Write success/failure report here and what you have tried.

In report, please include Windows version, user account type and Wireshark version (if applicable).

It is recommended to use USBPcapCMD as extcap. If you have recent Wireshark version, simply place the USBPcapCMD (the version to be tested) in C:\Program Files\Wireshark\extcap directory and restart Wireshark.

gpotter2, 2018-08-05:
@desowin Isn't there a mistake in the doc here: http://desowin.org/usbpcap/captureformat.html

> USBPCAP_TRANSFER_ISOCHRONOUS
> When function is equal to USBPCAP_TRANSFER_ISOCHRONOUS (0) the header type is USBPCAP_BUFFER_ISOCH_HEADER

I think it should be `When transfer ...` according to its doc

> transfer (offset 22) determines the transfer type and thus the header type. See below for details.

desowin, 2018-08-05:
@gpotter2  The SmartScreen is related to the fact that the automated build is signed using the test (self issued) certificate. SmartScreen is entirely related to the signature and not to the actual code.

This whole Authenticode, unlike some ISO certifications and/or CE certification, are only related to the fact that the certificate was bought from the expensive CA and used for signing. The only thing it really does verify is the the software comes from a individual/company that proved his identity to the Certificate Authority.

desowin, 2018-08-05:
@gpotter2  You are right about the "function"/"transfer" mistake. I have corrected it. By the way, if you find other problems on the website, please note that the website contents are available in the gh-pages branch (and thus it's possible to open Pull Requests with proposed changes).

gpotter2, 2018-08-11:
@desowin Hi ! I'm trying to setup scapy/USBPcap integration.

I have several issues:

- `.\USBPcapCMD.exe -A -o - -b 134217728` starts a new frame with interactive mode, ignoring what instructions I gave:
[image]

I need to be able to call USBPcapCMD.exe from a piped powershell then parse its output.

- How do I print the list of devices from a command that I could then parse ?

Thanks a lot for any help

desowin, 2018-08-12:
@gpotter2  You didn't provide the root hub parameter. In your case it should be "-d \\.\USBPcap1". By the way, I am used to see multiple root hubs no devices, it's interesting to know that Surface Pro 4 has just a single USB 3.0 root hub (did I properly guess from the list of devices?).

If you want a list of devices that you could parse, then take a look at the [extcap](https://wiki.wireshark.org/Development/Extcap) interface.

The command to get the list of devices would be: "--extcap-interface \\.\USBPcap1 --extcap-config"

gpotter2, 2018-08-12:
Thanks ! Will look into it

You guessed correctly :)

desowin, 2018-08-14:
USBPcap 1.2.0.4 is now officially released.

closed 2018-08-14 by desowin

gpotter2, 2018-08-14:
Thanks @desowin for your help !

If you want to have a look, [here](https://github.com/secdev/scapy/pull/1559) is how I implemented USBpcap in scapy.

Good luck with the project !