desowin/usbpcap #56: Does USBPcapCMD.exe need admin rights/privileges to perform USB packet capture

kirankumarg81, 2018-08-09:
Would like to know if USBPcapCMD.exe need admin rights/privileges to perform USB capture.
Platform: Windows-7
USBPcap version: 1.2.0.3

It is installed by Admin and executed by user in my case. Let me know if admin need to set any permissions on USBPcapCMD.exe file.

Thank you.

desowin, 2018-08-10:
The actual capture needs Elevated priviledges. There's no need to set any special permissions on the USBPcapCMD - if you start capture from unelevated USBPcapCMD, it'll automatically display the UAC elevation dialog.

The only reason for the admin rights is to prevent malicious use of the USBPcapDriver (eg. to prevent unelevated viruses from capturing the USB traffic).

desowin, 2018-08-10:
If you run USBPcapCMD as non-admin user, the UAC screen will ask for admin username and password.

desowin, 2018-08-14:
Closing this as this is basic security precaution to require admin rights for capturing on USB devices (USB keyboards are really common nowadays).

closed 2018-08-14 by desowin

kirankumarg81, 2018-08-14:
Thanks for the reply, it was very useful.