desowin/usbpcap #58: Update project website

desowin, 2018-08-14:
Update project website to cover the extcap Wireshark integration. Also include any and all important information that can be scattered across the github issues.

desowin, 2019-05-23:
Thanks to excellent contribution from @gpotter2 the gh-pages branch uses Jekyll to generate the pages.

bundito, 2019-10-10:
I am just this moment trying to grasp USBPCap and Wireshark to see what's going on between my RGB fan controller and the USB header it's connected to. I think I isolated the right USB adress, because Wireshark stopped reporting every USB keypress, and only showed data when I fooled with the LED controller app. What it means, I still have no idea.

ASIDE FROM NOT UNDERSTANDING, I'd be willing to do a little work on the website. It's a little, um, Y2K style. Although I used to be a web developer in a former life, I never learned all the flashy CSS stuff that the kids are doing these days. But this doesn't seem to be the kind of project that needs all that jazz. 

With a little guidance as to what things mean and how they should be organized, I might be able to get you a fresh coat of paint without massively disrupting a site many people have known for a long time.

Let me know.

desowin, 2019-10-11:
As you are struggling with the USBPcap right now, you know (or will know) what kind of information is missing from the website. The way you wrote the first paragraph suggests that you either didn't figure out the extcap integration at all or it is too complex.

Basically, since quite few years, Wireshark has feature called [extcap](https://www.wireshark.org/docs/man-pages/extcap.html). USBPcap was one of the first projects using that interface to integrate with Wireshark. Basically, you just copy the USBPcapCMD into wireshark/extcap directory. Then on next Wireshark start, you should see USBPcap interfaces in the interfaces list. Clicking on the gearbox symbol next to interface shall open the options, where you can select individual devices to capture. I believe that after you see that widget, it's clear enough what the individual options mean (unless the wording is not confusing).

I think the most important changes to the website would be new screenshots and updated tour. Keep the existing tour for reference/historical reasons (e.g. for the people who want to use USBPcap on Windows XP where the piping is the only way to get live capture). Basically, make the starting page dead obvious that USBPcap can be used directly from Wirehark GUI.

The website is available in git repository under the gh-pages branch. You can fork the repository, commit to your fork and then open the pull request here. I am happy to review (on the technical merit) any content changes you make.

desowin, 2020-10-28:
The website now contains Videos subpage where I linked USB Analysis 101. The video contains pretty much everything I would like to be present on the website - translating that into modern looking website requires some actual writing and web design skills.