desowin/usbpcap #60: 'Virtual' hub is not detected

MSLaaf, 2018-08-31:
HP  used hardware / software from EdiMax (and didn't release OSS for it - even though it shows linux) to redirect windows(only) usb traffic to a ScanJet N6350 scanner. It shows as a USB device in the system, but there is a software (virtual) driver installed simulating a HUB. I do not see it in USBPCap 1..4, yet it's clearly a USB driver.
They transport the data over UDP to/from the scanner, so it would be nice to see the actual USB traffic on the windows workstation and the resulting IP traffic...
Problem is that HP has NOT released SANE drivers to support this scanner on Linux.

Network Scanner Driver
- Device ID WIRELESS_USB\HPSS_EHCI\1&2D12BED1&0&01
- Status 0x0180600a Started
- Problem 0x00000000 (0)
- Service HPSSEHCD
- Capabilities 0x00000086
- Config Flags 0x00000000
- Class USB
- Manufacturer Edimax
- Hardware IDs Wireless_USB\HPSS_EHCI
- Compatible IDs Wireless_USB\HPSS_COMPATIBLE_EHCI
- Class GUID {36fc9e60-c465-11cf-8056-444553540000}
- Bus number 0x00000000
- Enumerator name WIRELESS_USB
- Description Network Scanner Driver
- Driver {36fc9e60-c465-11cf-8056-444553540000}\0030
- Physical Object Name \Device\0000025c
- UI number 0x00000001
- Bustype GUID {41203230-2037-4234-2035-412034372032}
- Legacy bus type 0x0000000f
- Install State 0x00000000
- Device Address 0x00000001
- Device Configuration File oem101.inf
- DevLoader *ntkern
- NTMPDriver hpssec.sys
- InfPathoem101.inf
- InfSection HPSSEHCD
- InfSectionExt .NT
- ProviderNameEdimax
- DriverDateData 00 C0 2F 9F CE 3F CA 01
- DriverDate 9-28-2009
- DriverVersion 1.0.0.2
- MatchingDeviceId wireless_usb\hpss_ehci
- DriverDesc Network Scanner Driver
- Class USB
- ClassDesc Universal Serial Bus controllers
- IconPath %systemroot%\system32\setupapi.dll,-20
- NoInstallClass1
- LowerLogoVersion 5.2
- UpperFilters USBPcap

HP Scanjet N6350
- Device ID USB\VID_03F0&PID_4805\CN54NEE04G05_N
- Status 0x0180600a Started
- Problem 0x00000000 (0)
- Service usbscan
- Capabilities 0x00000096
- Config Flags 0x00000000
- Class Image
- Manufacturer Hewlett-Packard
- Hardware IDs USB\Vid_03f0&Pid_4805&Rev_0604
- Hardware IDs USB\Vid_03f0&Pid_4805
- Compatible IDs USB\Class_ff&SubClass_ff&Prot_ff
- Compatible IDs USB\Class_ff&SubClass_ff
- Compatible IDs USB\Class_ff
- Class GUID {6bdd1fc6-810f-11d0-bec7-08002be2092f}
- Bus number 0x00000000
- Enumerator name USB
- Description HP Scanjet N6350
- Driver {6bdd1fc6-810f-11d0-bec7-08002be2092f}\0000
- Physical Object Name \Device\00000cb9
- UI number 0x00000001
- Bustype GUID {31203134-2043-3535-2036-322044312030}
- Legacy bus type 0x0000000f
- Install State 0x00000000
- Device Address 0x00000000
- Device Configuration File oem95.inf
- CoInstallers32hpn6350co.dll,HPScannerCoInstaller
- HardwareConfig 01
- DevLoader *NTKERN
- NTMPDriver usbscan.sys
- USDClass {79EE0032-4BEC-4595-8883-1335B740A34C}
- InfPathoem95.inf
- IncludedInfs sti.inf
- InfSection n6350.NTWIA
- ProviderNameHewlett-Packard
- DriverDateData 00 40 7C 15 A7 D9 C9 01
- DriverDate 5-21-2009
- DriverVersion 12.5.0.184
- MatchingDeviceId usb\vid_03f0&pid_4805
- DriverDesc HP Scanjet N6350
- CreateFileName \\.\Usbscan0
- SubClass StillImage
- Capabilities 49
- DeviceType 1
- DeviceSubType 4
- ICMProfiles sRGB Color Space Profile.icm
- FriendlyNameHP Scanjet N6350
- DeviceID {6BDD1FC6-810F-11D0-BEC7-08002BE2092F}\0000
- VendorHewlett-Packard
- IsPnP 1
- Class Image
- ClassDesc Imaging devices
- IconPath %systemroot%\system32\sti_ci.dll,-1000
- Installer32 sti_ci.dll,ClassInstall
- LowerLogoVersion 5.2
- UpperFilters ksthunk

I don't know enough about windows driver development to support this device. Is it possible to do so?

[N6350_Drivers.pdf](https://github.com/desowin/usbpcap/files/2341808/N6350_Drivers.pdf)

MSLaaf, 2018-09-01:
Output from USPPcapCMD:

following filter control devices are available:
 \\.\USBPcap4
 \??\USB#ASMEDIAROOT_Hub#5&7235832&1&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}
   [Port 3] USB Composite Device
     Logitech USB Input Device
       HID Keyboard Device
     USB Input Device
       HID-compliant mouse
       HID-compliant consumer control device
       HID-compliant device
       HID-compliant device
     Logitech Unifying USB receiver
       Logitech HID-compliant Unifying device
         HID-compliant device
         HID-compliant device
       Logitech HID-compliant Unifying device
         Logitech HID-compliant Unifying Mouse
           Logitech Driver Interface
           Logitech Driver Interface
       Logitech HID-compliant Unifying device
         Logitech HID-compliant Unifying keyboard
           Logitech Driver Interface
         HID-compliant consumer control device
         HID-compliant device
 \\.\USBPcap1
 \??\USB#ROOT_HUB20#4&f432ec7&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}
   [Port 1] Generic USB Hub
 \\.\USBPcap2
 \??\USB#ROOT_HUB20#4&144fac0d&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}
   [Port 1] Generic USB Hub
 \\.\USBPcap3
 \??\IUSB3#ROOT_HUB30#4&318a7e80&1#{f18a0e88-c30c-11d0-8815-00a0c906bed8}
   [Port 5] USB 2.0 MTT Hub
     [Port 3] USB Input Device
       HID-compliant device
   [Port 9] USB Input Device
     HID-compliant device
   [Port 11] USB Mass Storage Device
     XM-35U CF  PRO Reader USB Device
     XM-35U SD  PRO Reader USB Device
     XM-35U XD  PRO Reader USB Device
     XM-35U MS  PRO Reader USB Device
     XM-35U mSD PRO Reader USB Device
   [Port 12] Crystalfontz CFA635-USB
     Crystalfontz CFA635-USB (COM3)
   [Port 14] American Power Conversion USB UPS
     APC UPS
select filter to monitor (q to quit):

desowin, 2018-09-01:
Try starting elevated cmd prompt and run "USBPcapCMD -I" in it. If it find non standard HWID that looks like the virtual hub then it should be possible to capture from it after reboot.

If not, then try to find the hardware id of the virtual root hub in device manager. It might be easiest to sort the tree by connection (default is by type).

(Possible alternative: How are you physically connectin to he device? Is it via network?  If this is he case, it might be possible to write dissector for thus usb over ip implementation.)

desowin, 2018-09-01:
If the "Network Scanner Driver" is the root hub, then you even if `USBPcapCMD -I` does not find it, you might try to add the `Wireless_USB\HPSS_EHCI` to the `SYSTEM\CurrentControlSet\services\USBPcap\NonStandardHWIDs` REG_MULTI_SZ key and reboot.