desowin/usbpcap #62: Passthrough USB devices to VirtualBox doesn't work if USBPcap is installed

aholzinger, 2018-10-16:
If USBPcap is installed (I installed it together with Wireshark) VirtualBox passthrough of USB devices is not anymore working as expected.

This website explains it in detail:
http://www.zachpfeffer.com/single-post/A-fix-for-a-USB-drive-or-any-USB-device-not-working-in-VirtualBox

There is also an issue reported in VirtualBox:
https://www.virtualbox.org/ticket/16032

desowin, 2018-10-16:
While the VirtualBox issue mentions that it shouldn't be fixed in VirtualBox, is the actual root cause known? Last time I tried to investigate the issue, I failed to figure out what's wrong. Does anyone know if "fix in USBPcap" would need to be some specific workaround for VirtualBox or if there's really something broken in the USBPcap driver?

lorenzofattori, 2020-01-24:
Looks like this issue is still happening after many years.
One dirty solution is to remove the upperfilters in the registry settings and then virtualbox works.
Does USBpcap use the upperfilters in some way? Will it affect the behavior of USBpcap if I remove the upperfilters?

Trying to find a more broad workaround here

desowin, 2020-01-24:
> Does USBpcap use the upperfilters in some way? Will it affect the behavior of USBpcap if I remove the upperfilters?

USBPcap will no longer work if you remove it from UpperFilters. The way USBPcap captures packets is by attaching itself to the USB stack and looking at the URBs.

> Trying to find a more broad workaround here

I don't know any possible workaround. However, the potential solution is to hook to the USB stack using the same method as used in https://github.com/zodiacon/DriverMon/blob/master/DriverMonitor/DriverMon.cpp#L237
The reasons why USBPcap does not do it:
- I only heard about this method recently
- Even if I did it, I don't know how to translate the USB_PIPE_HANDLE into the actual address and endpoint numbers without capturing _URB_SELECT_CONFIGURATION. Currently USBPcap loads itself into the USB stack and looks at the URBs even if you don't capture. Without the capture active, USBPcap is only interested in _URB_SELECT_CONFIGURATION and uses the information there to build look up table with USB_PIPE_HANDLE as a key, and endpoint information in the value.
- The [usb3kd.ucx_endpoint](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/-usb3kd-ucx-endpoint) debugger command can give the required endpoint information but I don't know how and if it is possible to get such functionality using the API available to kernel drivers
- For USB 2 it should be possible somehow (SysNucleus USBTrace seems to be able to get the endpoint information given just the USB_PIPE_HANDLE; however it seems to be capable of doing it only for USB 2), but I have completely no idea how to do it

So if someone can provide information how to translate USB_PIPE_HANDLE in kernel mode to address and endpoint, for both USB 2 and 3 host controllers, then we could change the way USBPcap driver works. Such change would most likely also fix the #61.

MSoegtropIMC, 2021-07-18:
IMHO one major issue with USBPcap is that the uninstaller doesn't properly uninstall it. The following critical items remain after an uninstall of USBPCap (using Windows remove programs):

- File: `C:\Windows\System32\drivers\USBPcap.sys`
- Reg: `HKEY_LOCAL_MACHINE\SYSTEM\Setup\FirstBoot\Filters\{36fc9e60-c465-11cf-8056-444553540000}\UpperFilters`
- Reg: `Computer\HKEY_LOCAL_MACHINE\SYSTEM\Setup\FirstBoot\Services\USBPcap`

If the issue can't be fixed, at least the uninstaller should be fixed so that uninstalling USBpcap brings the system back to a working state.

desowin, 2021-07-18:
> IMHO one major issue with USBPcap is that the uninstaller doesn't properly uninstall it. The following critical items remain after an uninstall of USBPCap (using Windows remove programs):
>
> * File: [...]

These should be removed after a reboot. Did you check immediately after uninstalling or did you actually reboot?