desowin/usbpcap #64: Can't Filter By Device

MelbourneDeveloper, 2018-12-08:
I only want to trace this particular device (Port 11), but I only get the filter of "1". Is there some way to filter down to only port 11? I tried typing in 11 but I just get invalid input. When I ran the trace is dumped out 32 meg of data.

[image]

MelbourneDeveloper, 2018-12-09:
I went out and bought a USB hub because I thought I'd be able to filter:

[image]

But, it doesn't show up:

[image]

MelbourneDeveloper, 2018-12-09:
On top of this, it doesn't look like I can filter the outputted data in Wireshark by port:

[image]

MelbourneDeveloper, 2018-12-09:
I tried running wireshark like this:
USBPcapCMD.exe -d \\.\USBPcap2 -o - | "C:\Program Files\Wireshark\Wireshark.exe" -k -i -

But I just get this (no traffic)
[image]

MelbourneDeveloper, 2018-12-09:
I tried this

USBPcapCMD.exe -d \\.\USBPcap2 -o - | "C:\Program Files\Wireshark\Wireshark.exe" -k -i -"TREZOR Interface"

 but get this error:

[image]

MelbourneDeveloper, 2018-12-09:
I tried this 

"c:\program files\usbpcap\USBPcapCMD.exe" -d \\.\USBPcap2 -o - | "C:\Program Files\Wireshark\Wireshark.exe" -k -i -A

to get all devices but this error

[image]

MelbourneDeveloper, 2018-12-09:
Ok, it looks as though I was able to finally filter down the data when I found out the address. But, this is really clunky. Isn't there a better way? Have I done the right thing here? I still can't find the data I'm looking for. I think I've filtered out something that I need. 

[image]

MelbourneDeveloper, 2018-12-09:
Is there a way I can filter by a series of bytes like this:

0x3F
0x23
0x23

desowin, 2019-03-31:
Using the USBPcap from commandline catures the traffic from all devices connected to the selected root hub. If you want to capture the traffic for a single device, the recommented way is to use the USBPcapCMD.exe as an extcap. When using USBPcap as extcap, you can select the devices to capture on and start/stop the capture from the Wireshark GUI.

KOLANICH, 2019-05-27:
>When using USBPcap as extcap, you can select the devices to capture on and start/stop the capture from the Wireshark GUI.

This just doesn't work. At least for a USB mouse connected to the same hub.

desowin, 2019-05-28:
> > When using USBPcap as extcap, you can select the devices to capture on and start/stop the capture from the Wireshark GUI.
> 
> This just doesn't work. At least a USB mouse connected to the same hub floods into the capture even if an another device is captured.

Are you using Wireshark 3.0.2 and USBPcap 1.4.0.0? What are the commandline arguments Wireshark calls USBPcap with? Did you uncheck the capture from all devices option?

KOLANICH, 2019-06-18:
>Are you using Wireshark 3.0.2

yes.

>USBPcap 1.4.0.0

now 1.5.0.0 with the same effect

>What are the commandline arguments Wireshark calls USBPcap with?

```
USBPcapCMD.exe --capture --extcap-interface \\.\USBPcap1 --fifo \\.\pipe\wireshark_extcap_\\.\USBPcap1_20190618125930 --capture-from-all-devices --devices 2 --inject-descriptors --capture-from-new-devices
```

desowin, 2019-06-18:
> ```
> USBPcapCMD.exe --capture --extcap-interface \\.\USBPcap1 --fifo \\.\pipe\wireshark_extcap_\\.\USBPcap1_20190618125930 --capture-from-all-devices --devices 2 --inject-descriptors --capture-from-new-devices
> ```

This clearly shows that the "Capture from all devices connected" (tooltip: Capture from all devices connected despite other options) is set. When the option is not set, the "--capture-from-all-devices" is not present on the commandline. Either your system experiences some strange Wireshark bug (there were problems with extcap options, but these *should* be fixed in 3.0.2) or you have the checkbox ticked.

KOLANICH, 2019-06-18:
Thanks. That option really was enabled. Disabled that, worked as intended.

myjimmy, 2021-07-13:
@desowin 
Could you tell me how to use USBPcapCMD as extcap in more detail?

desowin, 2021-07-13:
> @desowin
> Could you tell me how to use USBPcapCMD as extcap in more detail?

Copy `USBPcapCMD.exe` to `C:\Program Files\Wireshark\extcap` then start Wireshark. USBPcap interfaces should then be visible in the interfaces list. Click on the icon next to the USBPcap interface to set capture options.

myjimmy, 2021-07-13:
@desowin Thanks.
I need to capture only a specific USB device like [this link](https://github.com/desowin/usbpcap/issues/64#issuecomment-503043138).
So, I run the following command:
```
USBPcapCMD.exe --capture --extcap-interface \\.\USBPcap1 --fifo \\.\pipe\wireshark_extcap_\\.\USBPcap1_20190618125930 --devices 2 --inject-descriptors --capture-from-new-devices
```
But I got the following error:
```
Thread started with invalid write handle!
```
[image]

I think that the pipe parameter `--fifo \\.\pipe\wireshark_extcap_\\.\USBPcap1_20190618125930` is wrong.
Please help.

myjimmy, 2021-07-14:
> Thanks. That option really was enabled. Disabled that, worked as intended.

@KOLANICH I need to capture the USB packets from a specific USB device.
So, I should use the following command provided by you:
```
USBPcapCMD.exe --capture --extcap-interface \\.\USBPcap1 --fifo \\.\pipe\wireshark_extcap_\\.\USBPcap1_20190618125930 --devices 2 --inject-descriptors --capture-from-new-devices
```
Could you let me know how to create the pipe like `--fifo \\.\pipe\wireshark_extcap_\\.\USBPcap1_20190618125930` in Wireshark?

desowin, 2021-07-14:
Wireshark creates the pipe using CreateNamedPipe function. Check extcap_create_pipe() in https://gitlab.com/wireshark/wireshark/-/blob/master/extcap.c#L1498

myjimmy, 2021-07-14:
@desowin Thanks for your reply.
To create the named pipe in Wireshark, shall I modify the source code?
Or which feature shall I use?

desowin, 2021-07-14:
> @desowin Thanks for your reply.
> To create the named pipe in Wireshark, shall I modify the source code?
> Or which feature shall I use?

Just start Wireshark, it should do everything. I thought you were writing your own application that invoked USBPcapCMD.

myjimmy, 2021-07-14:
@desowin Thanks.
When running `USBPcapCMD.exe` in the command prompt, I see several devices in `\\.\USBPcap1`.
[image]

**I need to capture a specific USB device with USBPcap.**

@KOLANICH described the method in [this link](https://github.com/desowin/usbpcap/issues/64#issuecomment-503043138).
But I got the following error:
```
Thread started with invalid write handle!
```
[image]

@desowin So, I hope you let me know how to capture a specific device with USBPcap.

pixeldoc2000, 2022-08-25:
@myjimmy 
Maybe it's easier the use to Filters in Wireshark like: `usb.src == "1.2.1"  or usb.dst == "1.2.1"` <- insert your USB Device Address