desowin/usbpcap #68: Device Manager - unknown device - USBpcap.sys

TomQuixote, 2019-01-16:
Hi. Since today I found an unknown device in the device manager. I don't know what to do now. Please help me.

Win 7 Prof. x64, Wireshark 2.6.6, USBPcap 1.2.0.4

[image]

[image]

[image]

desowin, 2019-01-16:
This is first time I see such issue. Is this device related to some external USB hub or is it internal to the computer? Can you try updating and/or uninstalling the driver in the Device Manager?

TomQuixote, 2019-01-18:
False alarm. It seems that a defective Bluetooth dongle caused several strange effects within the sysrem. Romving it an rebooting removed the unknown device. It appeared weirdly one time after rebooting but vanished. I don't know what this behaviour means.
Sorry for the hassle.