desowin/usbpcap #71: USBPcap error in windows10

Geniusssssss, 2019-02-11:
Detected unrecognized USB driver (\Driver\USBPcap).
I have uninstalled him, but I still get an error.

desowin, 2019-05-29:
Issue is not specific enough. The problem could be with other devices and USBPcap name appears because it is in UpperFilters.

ttimasdf, 2021-03-26:
It occurs to me the same. I get this warning every 5 seconds in event viewer. And Device Manager keeps refreshing the window like there's a device plugging in.

[image]

robinduerhager, 2022-02-05:
@desowin please reopen this issue.
I have this problem since months: reconnecting usb devices that do not exist so to say. Sometimes it was as bad as @ttimasdf says, today it's not occuring as frequently as it used to be.

ToasterDEV, 2023-03-14:
I have the same issue as described in this thread, if possible, please reopen this issue.

desowin, 2023-03-14:
Reopened but if you don't provide any real clues and/or traces there is nothing that can be done about it. I don't think this is USBPcap issue itself.

ToasterDEV, 2023-03-14:
> Reopened but if you don't provide any real clues and/or traces there is nothing that can be done about it. I don't think this is USBPcap issue itself.

First of all, thanks for the chance.

Now then, which logs and traces would be appropriate to begin troubleshooting?

At the moment I can say my USBPcap version is 1.5.4.0 running under Windows 11 Pro for Workstations Version 22H2 OS Build 22621.1344 (64-bit) on an Alienware m15 R6 (i7-11800H | 64GB DDR4 | RTX 3070 Laptop), but aside from the system's details not sure if providing an ADK trace would be a good place to begin or if I'm totally off-base.

desowin, 2023-03-14:
https://learn.microsoft.com/en-us/windows-hardware/drivers/devtest/irp-logging can give some insight but not sure if it'll really nail the problem.

Building debug (checked) version of the driver and reading the debug prints with DebugView can shed some light on what is happening.

desowin, 2023-03-14:
https://the-sz.com/products/rt/ is probably the easiest one to use with highest probability for giving out useful data.

ToasterDEV, 2023-03-20:
> [the-sz.com/products/rt](https://the-sz.com/products/rt/) is probably the easiest one to use with highest probability for giving out useful data.

First of all, thanks for the reference. While it took a lot of trying, I think I was able to get some useful output. I'll attach my trace in this comment. I did leave it running for about an hour, so perhaps the information collected ended up being too much, so if there's any better way for me to handle things, please let me know.

Thanks for the help!

https://1drv.ms/u/s!AlYW1BPO6zRtgYXCU8JJZpdAkzJGPtg?e=F9tfKJ

ntclon, 2023-06-14:
I had this error, only when having an Anker usb hub plugged in.  Updated realtek driver (the one popping in/out of device manager every few seconds) and it went away.  Not sure if this is helpful in this context, but thought I'd throw it in here since this was one of the few places the goog landed me, while searching my issue.

neilser, 2023-09-03:
> Pcap name appears because it is in UpperFilters.

If we remove it from the UpperFilters entry (or indeed remove _everything_ from it), would that give any more info about the root cause?

desowin, 2023-09-04:
> If we remove it from the UpperFilters entry (or indeed remove _everything_ from it), would that give any more info about the root cause?

I don't what what else you have in UpperFilters - blindly removing an entry other than USBPcap can render USB devices inoperable. Removing USBPcap alone from UpperFilters **and rebooting**, can completely rule out USBPcap involvement in the problem. That is, if you see the Unrecognized device error after USBPcap is removed from UpperFilters, then USBPcap has nothing to do with the issue.

neilser, 2023-09-04:
> I don't what what else you have in UpperFilters - blindly removing an entry other than USBPcap can render USB devices inoperable. Removing USBPcap alone from UpperFilters **and rebooting**, can [...]

Sorry, I didn't ask a very clear question; I'll try to rephrase it.

Since you indicate above (and elsewhere; I've seen a couple of issues in the project which touch on this topic) that this error is probably _incorrectly_ putting the blame on USBPcap, I am wondering if the underlying issue (whatever it may be) might be more readily identified after emptying the UpperFilters list.
In my own case, having just removed "nxusbh" from UpperFilters, USBPcap is now the only remaining entry.

I don't understand why USBPcap gets mentioned at all in these errors, but from what I've read here I'd be tempted to guess that it's simply because it's at the top of the stack. Is this actually the reason or is something else going on?

One might hope that after the filters are gone, the true culprit would become apparent but I fear that's not gonna happen...