desowin/usbpcap #80: Capture on PCIe MBIM card

janosvitok, 2019-05-17:
I'm trying to capture MBIM traffic from Fibocom L850-GL but USBPcap "does not see it". I suppose more cards with this architecture will appear in the near future.

Windows 10, 1809, 64bit.

The card's root device is attached to PCIe bus, and has "System" class. Under it there is normal USB 3 root hub (`USB3\ROOT_HUB30`) with USB class.
USBPcap filter is listed as driver for this root hub, and its ID is added to registry during `USBpcapcmd -I`.
The driver is also attached properly to all USB composite devices.

Problem is that there is no `\.\usbpcapN` device for this root hub (there are two devices for chipset's root hubs).

I suppose the problem is that usbpcap driver is not attached to root hub's parent, so it can't see when the root hub is added.

I tried to change inf file so that the driver is attached also to System devices, which prevented the machine to boot... :-)

What can be done to enable the capture from this device?

If you need any additional information or do any tests (I am able to compile driver from sources), let me know.

desowin, 2019-05-19:
Are you sure it is `USB3\ROOT_HUB30` and not `USB\ROOT_HUB30`?

Could you please confirm that `HKEY_LOCAL_MACHINE\SYSTEM\ControlSet001\Services\USBPcap` contains `NonStandardHWIDs` `REG_MULTI_SZ` with the root hub hardware ID?

Here is example from Windows 10 where the NonStandardHWIDs entry is correctly working:
[image]

If it is not there, the value to enter can be seen in Device Manager:
[image]

I think I should add the `USB\ROOT_HUB30` to the "standard" list, as Microsoft has official driver for it for quite some time.

janosvitok, 2019-05-19:
I am pretty sure it's USB3\ROOT_HUB30 and also that the ID is in the registry (I've checked sources to see how the devices are found, I hope I understood it correctly), I will double check ASAP when I get access to that device.

To save us one roundtrip, what additional info do you need when both are OK? (i.e. USB3\... and the ID is in the registry)? Dump some registry? Kernel dump? Device Manager content?

desowin, 2019-05-20:
The "Class Guid" Property value for the USB 3 Root Hub would be the next thing to check.

janosvitok, 2019-05-20:
I've checked the machine.

### Summary
- HWID is USB\ROOT_HUB30. I was wrong
- HWID is in the NonStandardHWIDs. I was correct.

Now something I forgot to mention or didn't state clearly: In the normal case the USB root hub's parent is USB controller with device class USB. In this case the controller is PCIe device and has class System. This is the only "physical", device, all others are "virtual", created by the controller's driver.

### Details

- **Device Manager screenshot**

[image]

- **HKEY_LOCAL_MACHINE\SYSTEM\ControlSet001\Enum dump**

[controlset001-enum.reg.txt](https://github.com/desowin/usbpcap/files/3198377/controlset001-enum.reg.txt)

- **Another dump of device properties**

[fibocom.txt](https://github.com/desowin/usbpcap/files/3198378/fibocom.txt)

- **UsbPcap registry screenshot**

[image]

- **UsbPcap registry dump**

[usbpcap.reg.txt](https://github.com/desowin/usbpcap/files/3198380/usbpcap.reg.txt)

- **NonStandardHWIDs content**

[NonStandardHWIDs.txt](https://github.com/desowin/usbpcap/files/3198379/NonStandardHWIDs.txt)

desowin, 2019-05-20:
Does bf759289284265ec5668ce192479a8e03d690c98 resolve the issue for you?

You have multiple entries in NonStandardHWIDs, and on my computer there was just one entry. I'll check if the code matching HWIDs handles that correctly.

janosvitok, 2019-05-20:
I removed all NonStandardHWIDs but the required one, and the capture port was not created either.

I suppose (knowing almost nothing about usbpcap) the problem is that there is no USB Controller above USB Root hub, so the usbpcap filter driver does not know when USB Root Hub is added, and no capture device is created.

As I wrote above, the immediate parent of USB Root Hub is a System device, and as such has no usbpcap filter driver attached.

janosvitok, 2019-05-20:
BTW. the IDs:
Device instance ID starts with USB\ROOT_HUB30, but Hardware ID starts indeed with USB3\ROOT_HUB30...

desowin, 2019-05-20:
> I tried to change inf file so that the driver is attached also to System devices, which prevented the machine to boot... :-)

How did you modify the inf file? Did your change do anything more than adding USBPcap to HKLM, System\CurrentControlSet\Control\Class\{4d36e97d-e325-11ce-bfc1-08002be10318} UpperFilters?

janosvitok, 2019-05-30:
I don't have the file anymore, but I believe that's what I did (add System class GUID). Sorry for the late response.