desowin/usbpcap #86: Capture more than 65535 bytes

MartyLake, 2019-08-04:
Hello !

I want to do a pipeline like `"C:\Program Files\USBPcap\USBPcapCMD.exe" --snaplen 134217728000 --bufferlen 134217728 --device \\.\USBPcap1 -A -o - |  python -u filter_interesting_packets.py | python -u display_real_time_visualization.py`

I am trying to capture the usb traffic for a device that uses bulk data transfer. Unfortunately, usbpcap only captures the beginning 65535 bytes for one packet.

I have read https://github.com/desowin/usbpcap/issues/26 and https://github.com/desowin/usbpcap/issues/22 and tried different version of usbpcap ( v1.4.1.0 https://groups.google.com/forum/#!searchin/usbpcap/snaplen|sort:date/usbpcap/6MuzIbIPh0g/3DNozJURBwAJ and  the latest 1.5.2.0 without success. I have tried to capture directly from wireshark, but I could not find the gui option there to set the snaplen (not relevant in my case since I want to do a realtime pipeline).

Do you have any idea why/what may I disable/enable on my system to make it capture everything ?

Best,

MartyLake, 2019-08-07:
I reproduced it on another "clean" computer, by installing only usbpcap 1.5.2.0.

desowin, 2019-08-07:
Using bigger snaplen than bufferlen does not make any sense - the packets won't be captured if they are bigger than buffer length. Another issue with your commandline option is using value bigger than (2^32)-1 as the snapshot length. Both usbpcap and the actual pcap format is limited to packets with length that can be stored in unsigned 32-bit unsigned integer.

MartyLake, 2019-08-07:
Hello @desowin , thanks for your answer.
I just tried with a shorter snaplen (500k) that is smaller than bufferlen (600k), both being smaller than 32bit unsigned integer max (4,294,967,295), and bigger than the size I want to capture (400k):

`C:\Program Files\USBPcap\USBPcapCMD.exe" --snaplen 500000 --bufferlen 600000 --device \\.\USBPcap1 -A -o - | python -u filter_interesting_packets.py | python -u display_real_time_visualization.py`

and it also reproduce :(
Do you know how I can debug this further ?

MartyLake, 2019-08-07:
I tried to follow the order of the arguments the way they are presented in the --help, without much success

`"C:\Program Files\USBPcap\USBPcapCMD.exe" --device \\.\USBPcap1 --output - --snaplen 500000 --bufferlen 600000 --capture-from-all-devices | python -u filter_interesting_packets.py | python -u display_real_time_visualization.py`

MartyLake, 2019-08-09:
I reproduced it on another "clean" computer, by installing only usbpcap 1.5.2.0, and using the sensible arguments.

`C:\Program Files\USBPcap\USBPcapCMD.exe" --snaplen 500000 --bufferlen 600000 --device \\.\USBPcap1 -A -o - | python -u filter_interesting_packets.py | python -u display_real_time_visualization.py`

desowin, 2019-08-11:
It looks that `generate_worker_command_line()` does not pass the snaplen to elevated worker. Does it work properly if you type the command inside an elevated command line? If not, then there's most likely some other bug with the snaplen handling.

MartyLake, 2019-08-11:
It seems like this workaround works !!!

closed 2019-08-11 by desowin

MartyLake, 2019-08-11:
Thank you very much !

hxqqqqqq, 2023-03-10:
Why can I see packets when I use it to monitor packet capturing, but the packets captured by the content serial port tool are completely different, and there is no inclusion relationship. Is there a good person who can answer questions?

desowin, 2023-03-10:
> Why can I see packets when I use it to monitor packet capturing, but the packets captured by the content serial port tool are completely different, and there is no inclusion relationship. Is there a good person who can answer questions?

How is this related to this issue? What do you mean by completely different? Note that "packets" captured by USBPcap are essentially URBs which contain metadata. The actual serial port payload should be included in the capture, but it will be surrounded by rather significant amount of metadata.

commit 987b6935 (2019-08-11): Pass snapshot length to elevated worker

Fixes: #86