desowin/usbpcap #88: question: how to use this to control mouse click in windows

xoraxrax, 2019-09-26:
i already use wireshark capture this packet：
[image]

i wanna use c/c++ api sent this packet again now.

is there any examples to test this idea?

best regards.

xoraxrax, 2019-09-26:
such as that the api like this :send_usb_packet(handle,packet,packetLen)

whc2001, 2019-09-29:
I don't think there is any API to send USB packet to system. Every packet is sent by a USB device, so you can either write your own virtual driver which I don't know how to, or use a microcontroller that has USB device hardware resource to do. Both way are not easy.

desowin, 2019-10-17:
@whc2001 is correct. To add on the technical side, the packet captured was sent from device to host. It was interrupt type, and thus most likely it was USB HID device (title suggest it was from USB mouse). The HID reports (interrupt transfers) are actually meaningless without accompanying USB HID Report Descriptors that are sent from device to host during enumeration.

https://stackoverflow.com/questions/47942366/windows-virtual-mouse-driver contains some information how to achieve what @xoraxrax wants.

closed 2019-10-17 by desowin