desowin/usbpcap #92: How to identify the USB Ports on USBPcapCMD ?

mg-ramezani, 2019-11-14:
I read some issues about 'how to only capture from special port ?', My problem is about 'how to identify the ports ?'.   
When i open the USBPcapCMD.exe it list filters and ports with a port number, But when i send those port number as `address_list` to USBPcapCMD.exe it's does't work as expected, I must try ports number to figure out which port number is my expected device.  For example it will show my Mouse port as 1, But when i send 1 as port number it start capture from my Keyboard.  
How to get right device port number from USBPcapCMD ?

desowin, 2019-11-14:
What do you mean by `address_list`? Are you modifying USBPcapCMD code? The address list is supposed to be passed by `--devices` commandline argument.

The filter operates on the *addresses* not on *ports*. You can get the addresses in device manager or using USBPcapCMD extcap calls. Example output of `USBPcapCMD.exe --extcap-interface \\.\USBPcap1 --extcap-config` is:
```
arg {number=0}{call=--snaplen}{display=Snapshot length}{tooltip=Snapshot length}{type=unsigned}{default=65535}
arg {number=1}{call=--bufferlen}{display=Capture buffer length}{tooltip=USBPcap kernel-mode capture buffer length in bytes}{type=integer}{range=0,134217728}{default=1048576}
arg {number=2}{call=--capture-from-all-devices}{display=Capture from all devices connected}{tooltip=Capture from all devices connected despite other options}{type=boolflag}{default=true}
arg {number=3}{call=--capture-from-new-devices}{display=Capture from newly connected devices}{tooltip=Automatically start capture on all newly connected devices}{type=boolflag}{default=true}
arg {number=4}{call=--inject-descriptors}{display=Inject already connected devices descriptors into capture data}{type=boolflag}{default=true}
arg {number=99}{call=--devices}{display=Attached USB Devices}{tooltip=Select individual devices to capture from}{type=multicheck}
value {arg=99}{value=3}{display=[3] USB Composite Device}{enabled=true}
value {arg=99}{value=3_1}{display=JLink CDC UART Port (COM6)}{enabled=false}{parent=3}
value {arg=99}{value=3_2}{display=J-Link driver}{enabled=false}{parent=3}
value {arg=99}{value=2}{display=[2] USB Composite Device}{enabled=true}
value {arg=99}{value=2_1}{display=Integrated Webcam}{enabled=false}{parent=2}
value {arg=99}{value=1}{display=[1] Intel(R) Wireless Bluetooth(R)}{enabled=true}
value {arg=99}{value=1_1}{display=Microsoft Bluetooth LE Enumerator}{enabled=false}{parent=1}
value {arg=99}{value=1_2}{display=Bluetooth Device (RFCOMM Protocol TDI)}{enabled=false}{parent=1}
value {arg=99}{value=1_3}{display=Microsoft Bluetooth Enumerator}{enabled=false}{parent=1}
value {arg=99}{value=1_4}{display=Bluetooth Device (Personal Area Network)}{enabled=false}{parent=1}
```

The entries that have `{enabled=true}` contain `value` that is valid address. The entries for composite devices are provided only to make it easier to determine what the "USB Composite Device" is.

mg-ramezani, 2019-11-14:
I mean `thread_data::address_list`, Yes i modifying USBPcapCMD.exe code.

I don't understand which number in above output must send as port number to `address_list` ?

desowin, 2019-11-14:
> I don't understand which number in above output must send as port number to `address_list` ?

If you want to capture from J-Link, then the value is `3`. If you wanted to capture from JLink and the Webcam, then it would be `3,2`. From Webcam only - `2`.