desowin/usbpcap #96: Why isn't USB Pcap capturing my video anymore?

jani127, 2020-01-30:
About a year ago I connected Rifle scopes to my computer's USB port.  Then, I invoked USBPcapCMD Application.   Pcap program captured the video from device.  Filename:  year_ago.PNG shows the bulk video 512 byte packets.

I still have the same pcap application but last month I got a new computer.  Today, pcap program isn't capturing the video packets from my Rifle scope USB device.  Why?  My USB device is correctly shown as USB Composite Device, BAE STS Bulk, USB CDC serial port for BAE STS (COM10).  Please see attached filename:  invoke_pcap.PNG.  But the Wireshark capture doesn't show the video packets.

USBPcapCMD file version is:  1.2.0.4

Rifle Scope device is sending the video because I can see live video in another windows program.  Why isn't usb packet capture program detecting video anymore?

[image]

[image]

desowin, 2020-01-31:
USBPcapDriver version is far more important than the userspace tool (USBPcapCMD). Could you please check different releases to see which one introduced the problem?

desowin, 2020-01-31:
It is important to note that in USB device cannot send anything unless host asks for it. Are you running the capture while the Rifle Scope application is displaying video on your computer?

The capture from year ago seems to have used very old USBPcapDriver. With recent versions the bulk URBs will be captured both when travelling down the stack and back up (regardless of payload direction).

jani127, 2020-02-02:
>>  Could you please check different releases to see which one introduced the problem?
I want to know why the release installed in my computer worked before and isn't working now?

>> Are you running the capture while the Rifle Scope application is displaying video on your computer?
No.  This must be the problem.  I'll start packet capture while host requesting video from the device.
I'll retry on Feb. 10th when I get back to office.
Thank you for finding the problem.

desowin, 2020-02-02:
> > > Could you please check different releases to see which one introduced the problem?
> > > I want to know why the release installed in my computer worked before and isn't working now?

The capture you showed the screenshot for was created by fairly old version. There have been multiple bugs fixed since then, and with any bug fix there's a slight possibility for a regression.

> > > Are you running the capture while the Rifle Scope application is displaying video on your computer?
> > > No.  This must be the problem.  I'll start packet capture while host requesting video from the device.
> > > I'll retry on Feb. 10th when I get back to office.
> > > Thank you for finding the problem.

So this is most likely just a user error (expecting URB when there isn't any) not an actual bug.

jani127, 2020-02-10:
Yes, I confirmed it was user error.  Now, pcap program is capturing video from the USB Device.

closed 2020-02-10 by jani127