desowin/usbpcap #100: USBPcapCMD could not monitor and capture USB port successfully

Vincent-hu903, 2020-05-15:
Hi Sir,

I met a strange issue that USBPcapCMD could not capture USB port successfully on one testing computer, but it could run well on other computer and notebooks.  After type target test.pcap file then run the capture process, it work abnormal.  I open the test.pcap file and found it seems USB enum failed and not capture any data never.   In my understanding, even though USB Device not be connected, the USB Port should has the cyclic data as heartbeat.

Please see attached zip file.

[USBPcapCMD capture error.zip](https://github.com/desowin/usbpcap/files/4633130/USBPcapCMD.capture.error.zip)

Thanks.

Best Regards,
Vincent

desowin, 2020-05-15:
> I met a strange issue that USBPcapCMD could not capture USB port successfully on one testing computer, but it could run well on other computer and notebooks. After type target test.pcap file then run the capture process, it work abnormal.

It looks like USBPcapCMD could not open the capture device. One possible cause can be another capture process being still active. You could "End task" on all USBPcapCMD processess in Task Manager and check if you can reproduce the issue.

> I open the test.pcap file and found it seems USB enum failed and not capture any data never. In my understanding, even though USB Device not be connected, the USB Port should has the cyclic data as heartbeat.

This is outright wrong. If there's no device connected, there's no communication on the port. When there is device connected then there is periodic signalling (if that signalling ceases, device enters suspend), but this signalling is on the Link-Layer. USBPcap captures the URBs, which is higher layer and thus is unable to show low level signalling. URBs are device driver requests to USB host controller.

Vincent-hu903, 2020-05-18:
Hi desowin,

Thanks for your support.

It was the 1st time open the USBPcapCMD.exe when occurred capture failed, so maybe it not caused by the other capture process being active.  But I could trace it later.

About the 2nd point you described,  maybe my previous description was not very clearly.  My meaning is even though there is no USB device ( such as mouse, keyboard, user usb device as HID, etc. ) be connected, the capture process still be active and run continuous with capture URB_BLUK in data.   But there is no active URB_BLUK in data be captured continuous when my issue occurred, so I suspect if there is Hardware configuration of PC BIOS or Win10 system issues ?   

Attached is USBPcap capture file on my notebook with no USB device connected.
[noUSBDeviceConnect.zip](https://github.com/desowin/usbpcap/files/4641737/noUSBDeviceConnect.zip)

Hope your response ASAP.

Thanks.

Best Regards,
Vincent

margartiaville, 2020-05-18:
I met the same issue.
Any conclusion?

desowin, 2020-05-18:
> About the 2nd point you described, maybe my previous description was not very clearly. My meaning is even though there is no USB device ( such as mouse, keyboard, user usb device as HID, etc. ) be [...]

Which USBPcapCMD version was used to create this capture?

Vincent-hu903, 2020-05-18:
DriverVer=08/20/2017,1.2.0.2

desowin, 2020-05-18:
> DriverVer=08/20/2017,1.2.0.2

Always use the latest version (currently 1.5.3.0) before reporting issues. Old versions are not supported.

Vincent-hu903, 2020-05-19:
> DriverVer=08/20/2017,1.2.0.2
> 
> Always use the latest version (currently 1.5.3.0) before reporting issues. Old versions are not supported.

Hi desowin,

Already installed 1.5.3.0 before and the same issue occurred, it's very strange that there are only 3 PCs has this problem, other PCs and notebooks are work well ( the old version 1.2.0.2 has the same behaviour). 

BTW: the 3 error PCs are the same DELL type with other PCs except PC bootloader version, so I doubt if this error caused by WIN10 system or PC hardware configuration in BIOS?  As I know, WIN10 improved many USB power configuration in different win10 version.

desowin, 2020-05-19:
You should recreate noUSBDeviceConnect.zip with latest USBPcap version.

Vincent-hu903, 2020-05-19:
I found the root cause .  Local control PC is no problem and remote control PC has this issue.