desowin/usbpcap #103: Is it possible to build the code in Win10 via NMake2MSBuild

royyy83512, 2020-06-15:
Hi,
I have asked #102 a couple days ago.
Because I cannot find a proper tool on VS2019 on Win10,
I tried to use NMake2MSBuild on command prompt VS2013 with WDK8.1.

It then successfully created dirs.sln and other related files, 
however, when I build dirs.sln with VS2013, most of the files have the errors and warning as below:

error C2220: warning treated as error - no 'object' file generated
warning C4819: The file contains a character that cannot be represented in the current code page (950). Save the file in Unicode format to prevent data loss

Any suggestions? Thanks again.

royyy83512, 2020-06-15:
The messages:

1>------ Build started: Project: USBPcap, Configuration: Win8.1 Debug Win32 ------
2>------ Build started: Project: USBPcapCMD (USBPcapCMD\USBPcapCMD), Configuration: Win8.1 Debug Win32 ------
1>  Stamping .\Win8.1Debug\x86\\USBPcap.inf [Version] section with DriverVer=06/15/2020,1.5.4.0
2>  cmd.c
2>cmd.c : error C2220: warning treated as error - no 'object' file generated
2>cmd.c : warning C4819: The file contains a character that cannot be represented in the current code page (950). Save the file in Unicode format to prevent data loss
1>  usbpcapbuffer.c
1>usbpcapbuffer.c : error C2220: warning treated as error - no 'object' file generated
1>usbpcapbuffer.c : warning C4819: The file contains a character that cannot be represented in the current code page (950). Save the file in Unicode format to prevent data loss
1>c:\users\rluo1\desktop\usbpcap\usbpcapdriver\USBPcapMain.h : warning C4819: The file contains a character that cannot be represented in the current code page (950). Save the file in Unicode format to prevent data loss
1>c:\users\rluo1\desktop\usbpcap\usbpcapdriver\USBPcapQueue.h : warning C4819: The file contains a character that cannot be represented in the current code page (950). Save the file in Unicode format to prevent data loss
1>c:\users\rluo1\desktop\usbpcap\usbpcapdriver\include\USBPcap.h : warning C4819: The file contains a character that cannot be represented in the current code page (950). Save the file in Unicode format to prevent data loss
1>c:\users\rluo1\desktop\usbpcap\usbpcapdriver\USBPcapBuffer.h : warning C4819: The file contains a character that cannot be represented in the current code page (950). Save the file in Unicode format to prevent data loss
1>c:\users\rluo1\desktop\usbpcap\usbpcapdriver\USBPcapHelperFunctions.h : warning C4819: The file contains a character that cannot be represented in the current code page (950). Save the file in Unicode format to prevent data loss
1>  usbpcapdevicecontrol.c
1>usbpcapdevicecontrol.c : error C2220: warning treated as error - no 'object' file generated
1>usbpcapdevicecontrol.c : warning C4819: The file contains a character that cannot be represented in the current code page (950). Save the file in Unicode format to prevent data loss
2>c:\users\rluo1\desktop\usbpcap\usbpcapcmd\filters.h : warning C4819: The file contains a character that cannot be represented in the current code page (950). Save the file in Unicode format to prevent data loss
2>c:\users\rluo1\desktop\usbpcap\usbpcapcmd\thread.h : warning C4819: The file contains a character that cannot be represented in the current code page (950). Save the file in Unicode format to prevent data loss
2>C:\Users\rluo1\Desktop\usbpcap\USBPcapDriver\include\USBPcap.h : warning C4819: The file contains a character that cannot be represented in the current code page (950). Save the file in Unicode format to prevent data loss
2>c:\users\rluo1\desktop\usbpcap\usbpcapcmd\enum.h : warning C4819: The file contains a character that cannot be represented in the current code page (950). Save the file in Unicode format to prevent data loss
2>c:\users\rluo1\desktop\usbpcap\usbpcapcmd\roothubs.h : warning C4819: The file contains a character that cannot be represented in the current code page (950). Save the file in Unicode format to prevent data loss
2>c:\users\rluo1\desktop\usbpcap\usbpcapcmd\version.h : warning C4819: The file contains a character that cannot be represented in the current code page (950). Save the file in Unicode format to prevent data loss
2>c:\users\rluo1\desktop\usbpcap\usbpcapcmd\descriptors.h : warning C4819: The file contains a character that cannot be represented in the current code page (950). Save the file in Unicode format to prevent data loss
2>c:\users\rluo1\desktop\usbpcap\usbpcapcmd\iocontrol.h : warning C4819: The file contains a character that cannot be represented in the current code page (950). Save the file in Unicode format to prevent data loss
2>  descriptors.c
1>c:\users\rluo1\desktop\usbpcap\usbpcapdriver\USBPcapMain.h : warning C4819: The file contains a character that cannot be represented in the current code page (950). Save the file in Unicode format to prevent data loss
2>descriptors.c : error C2220: warning treated as error - no 'object' file generated
2>descriptors.c : warning C4819: The file contains a character that cannot be represented in the current code page (950). Save the file in Unicode format to prevent data loss
1>c:\users\rluo1\desktop\usbpcap\usbpcapdriver\USBPcapQueue.h : warning C4819: The file contains a character that cannot be represented in the current code page (950). Save the file in Unicode format to prevent data loss
1>c:\users\rluo1\desktop\usbpcap\usbpcapdriver\include\USBPcap.h : warning C4819: The file contains a character that cannot be represented in the current code page (950). Save the file in Unicode format to prevent data loss
1>c:\users\rluo1\desktop\usbpcap\usbpcapdriver\USBPcapURB.h : warning C4819: The file contains a character that cannot be represented in the current code page (950). Save the file in Unicode format to prevent data loss
1>c:\users\rluo1\desktop\usbpcap\usbpcapdriver\USBPcapRootHubControl.h : warning C4819: The file contains a character that cannot be represented in the current code page (950). Save the file in Unicode format to prevent data loss
[... cut ...]
1>c:\users\rluo1\desktop\usbpcap\usbpcapdriver\USBPcapHelperFunctions.h : warning C4819: The file contains a character that cannot be represented in the current code page (950). Save the file in Unicode format to prevent data loss
1>  Generating Code...
2>  Generating Code...
3>------ Build started: Project: dirs-Package (Dirs-Package\dirs-Package), Configuration: Win8.1 Debug Win32 ------
3>C:\Program Files (x86)\Windows Kits\8.1\build\WindowsDriver.common.targets(1366,5): error MSB3030: Could not copy the file "C:\Users\rluo1\Desktop\usbpcap\USBPcapDriver\Win8.1Debug\x86\USBPcap.sys" because it was not found.
========== Build: 0 succeeded, 3 failed, 0 up-to-date, 0 skipped ==========

desowin, 2020-06-15:
This is one of the reasons why #87 has the "help wanted" label.

The next issues you will observe will be related to DefaultUninstall section. See [Creating a primitive driver](https://docs.microsoft.com/en-us/windows-hardware/drivers/develop/creating-a-primitive-driver) for more information.

Then, there is the issue how to build driver for Windows 7 and Windows 8. There are few parts that use preprocessor `#if` on `NTDDI_VERSION` to conditionally compile code for older or newer Windows version. This should either be changed to create multi-version driver, or create separate configurations for Windows 7 and Windows 8 (so far no Windows 10 specific functions are being used).

Ideally, the Visual Studio solution files would reside in repository top-level `vs2019` directory. The `.bat` files uses to create release should be updated to use it.