desowin/usbpcap #104: Not allowing to redirect output of USBPcapCMD.exe.

Huded14, 2020-06-20:


desowin, 2020-06-20:
I don't understand what you mean by "Not allowing to redirect output of USBPcapCMD.exe" in title. Just specify "-" as the output and USBPcapCMD will output the pcap stream to stdout. For a finer control you might use the extcap interface.

Huded14, 2020-06-22:
output of USBPcapCMD.exe
[image]

I want to redirect the console to text file

desowin, 2020-06-22:
The recommended way is to use extcap interface in such cases. It is suited for machine processing.

Use `USBPcapCMD.exe --extcap-interfaces` to obtain list of available interfaces, the example output would be:
```
interface {value=\\.\USBPcap1}{display=USBPcap1}
interface {value=\\.\USBPcap2}{display=USBPcap2}
interface {value=\\.\USBPcap3}{display=USBPcap3}
```

Then to get information about devices connected, use e.g. `USBPcapCMD.exe --extcap-config --extcap-interface \\.\USBPcap3`. This will list all available commandline options, and also all devices. It allows you to select individual devices (it is not possible using the console interface you refer to), using the `--devices` option.

Huded14, 2020-06-22:
I want to retrieve which device is connected to which interface (USBPcap? i.e., USBPcap1 or USBPcap2 or USBPcap3....)
EX: suppose i have connected device Sandisk Ultra USB Device which has detected in USBPcap3 .

To get such info what i should use ?

desowin, 2020-06-22:
> I want to retrieve which device is connected to which interface (USBPcap? i.e., USBPcap1 or USBPcap2 or USBPcap3....)
> EX: suppose i have connected device Sandisk Ultra USB Device which has detected in USBPcap3 .
> 
> To get such info what i should use ?

`USBPcapCMD.exe --extcap-config --extcap-interface \\.\USBPcap3`