desowin/usbpcap #106: USBpcap causes BSOD on Windows startup every time

qris, 2020-08-26:
The computer has rebooted from a bugcheck.  The bugcheck was: 0x0000007e (0xffffffffc0000005, 0xfffff80368499e95, 0xffffbf0b9e924f28, 0xffffbf0b9e924770). A dump was saved in: C:\WINDOWS\MEMORY.DMP. Report Id: 981702ba-1fd6-494f-bb26-3f5a4ee80e4d.

This happened immediately after installing USBpcap. After a number of failed boot attempts, the system repaired itself (reverted to the restore point created during installation) which removed the USBpcap driver, and it then boots again.

This has happened on both my attempts to install USBpcap on this box.

desowin, 2020-08-27:
Please send me the memory dump (can be privately to my email). What Windows version are you using?

qris, 2020-08-27:
Sent by email, thanks :) Windows 10 Pro, 64-bit.