desowin/usbpcap #107: Not listing usb devices shared via network

realivanjx, 2020-10-07:
Hi,

I have devices that are shared via network (with usb redirector technician edition). It works well with my programs with said device just like it is directly connected to my pc. But unfortunately usbpcap does not list it. Here are what it says:

```
Following filter control devices are available:
1 \\.\USBPcap1
  \??\USB#ROOT_HUB30#5&5b2d6c9&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}
    [Port 7] USB Input Device
      HID-compliant mouse
    [Port 8] USB Composite Device
      USB Input Device
        HID Keyboard Device
      USB Input Device
        HID Keyboard Device
        HID-compliant system controller
        HID-compliant consumer control device
        HID-compliant vendor-defined device
    [Port 12] Xbox One Controller
      USB Input Device
        XINPUT compatible HID device
2 \\.\USBPcap2
  \??\USB#ROOT_HUB30#5&1888fbf0&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}
3 \\.\USBPcap3
  \??\USB#ROOT_HUB30#5&ab90873&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}
Select filter to monitor (q to quit):
```

and with `-I`:

```
Hardware ID: USB\ROOT_HUB30&VID1022&PID15E1
Hardware ID: USB\ROOT_HUB30
Found standard HWID
Hardware ID: USB\ROOT_HUB30&VID1022&PID15E0&REV0000
Hardware ID: USB\ROOT_HUB30&VID1022&PID15E0
Hardware ID: USB\ROOT_HUB30
Found standard HWID
Hardware ID: USB\ROOT_HUB30&VID1022&PID43BC&REV0002
Hardware ID: USB\ROOT_HUB30&VID1022&PID43BC
Hardware ID: USB\ROOT_HUB30
Found standard HWID
```

Here is my device:
[image]

Any help is appreciated.
Thanks.

realivanjx, 2020-10-08:
This also might be related to #97

nefarius, 2022-05-20:
This is unfortunately "by design" and not fixable by USBPcap. These virtual USB emulation drivers don't emulate a host controller and a hub, they're basically bare-minimal bus drivers that just expose PDOs with USB device interfaces and handle URBs. Without a proper hub device object, USBPcap can not enumerate and load onto them, hence not being available for capture.

[image]

realivanjx, 2022-05-20:
im closing this as this is not the issue with usbpcap

closed 2022-05-20 by realivanjx

nefarius, 2022-05-20:
> im closing this as this is not the issue with usbpcap

Same case with #97 👌