desowin/usbpcap #108: capture serial data using USBPcap

BMWE, 2020-10-09:
Hi,

Few weeks ago I've made some test: Using USBPcap and I've used to capture RS232 and RS422 data. I've used RS to USB convertor.

For RS232, the data was textual. Each row that I've seen on the terminal was as single frame on the wireshark.
For RS422, the data was binary. Each byte that was sent, was as single frame in the wireshark.
Any idea how to solve the issue for the RS422 so that the whole message would be as single frame?

(If I'm not wrong for RS422->USB I've used the ATEN adapter. No idea which one was used for the RS232).

desowin, 2020-10-09:
To achieve what you want you would have to either change the serial converter driver and/or application. USBPcap itself cannot really influence the traffic in any way.

If the converter is FTDI FT based one:
- you could try increasing latency timer to group more RX bytes into one packet. Note that this make break the communication if the application and/or target device is not tolerant enough.
- with recent Wireshark you can write dissector on top of FTDI FT dissector and request more data using the desegment API (FTDI FT implements the one more segment option using desegment API).
- for TX data, it is determined by how the application writes the data. If application writes one byte at a time, then there will be one URB per each byte.

If the converter uses some different chip, you could probably write Wireshark dissector for it.

asulwer, 2025-10-05:
what are the steps necessary to duplicate this? i need to see the data or filter out everything except the data to/from my device