desowin/usbpcap #110: Understanding,Filtering,sorting kernel and “real”data

eljws, 2020-10-21:
Greetings I’m a very beginner with USB, embedded and I’m struggling to get my head around the protocol. When using wire shark with USB Mouse I think it gives me a lot more information that I need for my learning which Is a bit confusing.
For example irp id, which I read are part of kernel. It’s usefull for learning about operating systems but I would prefer to hide this. Is there a way to do that. Or up to which point kernel data is printed and actual data starts? What should I look for? Thanks

desowin, 2020-10-26:
I think I have covered all things beginners need during SharkFest'20 Virtual in the USB Analysis 101 presentation. The slides are available at the [retrospective page](https://sharkfestvirtual.wireshark.org/sf20v) and the recording is at [YouTube](https://www.youtube.com/watch?v=cUljKImph4s).

eljws, 2020-10-26:
Wow that's like the best tutorial on USB I have ever seen. 

It helped me understand what some of knowledge I seen in books or USB specs was to do with PHY layer and some where in layers above. Now I can organise it in my head what information belongs where. 

The OS metadata for drivers is no interest of me. So I will totally ignore the URB section of messages.

I enjoyed the recommendations for PHY layer debugging on the budget tool.