desowin/usbpcap #117: EV Code Signing Certificate for open source

Slion, 2021-04-19:
@desowin Just trying to contact you about kernel mode signing as I figure you went through that yourself.
First off though thanks for USBPcap it's really awesome to [reverse engineer HID traffic](https://github.com/Slion/TrackPointDriver/blob/d4d651f31ec29a5fe10dac9c36c3c3596778ca93/docs/trackpoint-sensitivity-report.md). I may need to put it to good use sooner or later.

I'm looking for ways to get [Slion/TrackPointDriver](https://github.com/Slion/TrackPointDriver) signed.
Would you be happy to sign it yourself with your certificate in exchange for a donation?

If that's no option for you, do you have some advise on how to get this done?
From where did you buy your EV certificate? At which cost?
It looks like EV certificates are meant to be issued to company rather than individuals, yet yours was issued to your name it seems. What's the trick to get this done correctly?

desowin, 2021-04-19:
> I'm looking for ways to get [Slion/TrackPointDriver](https://github.com/Slion/TrackPointDriver) signed.
> Would you be happy to sign it yourself with your certificate in exchange for a donation?

Unfortunately that's not an option as it involves company EV certificate.

> If that's no option for you, do you have some advise on how to get this done?

Investigate whether it is possible to convert the driver to UMDF.

> From where did you buy your EV certificate? At which cost?

Always make sure to check the latest requirements regarding Attestation Signing before you buy the certificate. The "allowed CA" used to change over time.

> It looks like EV certificates are meant to be issued to company rather than individuals, yet yours was issued to your name it seems. What's the trick to get this done correctly?

The trick is that there are two certificates involved. The Standard certificate issued by Certum that is used to sign the installer and drivers for all Windows versions except 10, and a EV company certificate used for Attestation Signing.

closed 2021-04-19 by desowin

Slion, 2021-04-21:
That was worth a try. Thanks for the tips.