desowin/usbpcap #119: USB Audio packets

robcazzaro, 2021-12-21:
I'm sure it's a PBKAC and not a real issue (Problem Between Keyboard And Chair... i.e. me :) )

I've used USBPcap and Wireshark before to capture USB HID packets, and it worked well. I just installed the latest Wireshark and USBPcap on my Windows 10 laptop (Wireshark 3.6, USBPcap 1.5.4.0 and Win 10 19044.1415)

I have a Topping USB DAC/Amplifier (MX3) that works well when connected directly to the laptop. When going thru a USB hub, the device is recognized but doesn't play any sound. It works with a different USB hub, so it's something specific to that one hub. 

With Wireshark and USBPcap I can see the descriptor packets and URB packets for things like changing the volume. But I cannot see any actual audio packets.

I listened to your talk and I think that only the URB packets are captured, not to isochronous audio ones... but I saw from Google searches that you actually fixed a isochronous packets issue, albeit for IN packets. https://github.com/desowin/usbpcap/issues/21

Is there a way to see the OUT audio packets? If not, my only other option is to use a logic analyzer with sigrok and the USB packet decoder, which is much less convenient than Wireshark

desowin, 2021-12-21:
> I have a Topping USB DAC/Amplifier (MX3) that works well when connected directly to the laptop. When going thru a USB hub, the device is recognized but doesn't play any sound. It works with a different USB hub, so it's something specific to that one hub.

Is one of the hubs USB2 and the other USB3? If so, with which one does it work and which one it doesn't? Are both hubs bus powered or do they have their own power supply?

> With Wireshark and USBPcap I can see the descriptor packets and URB packets for things like changing the volume. But I cannot see any actual audio packets.

You won't see actual audio packets but you should see isochronous URBs.

> I listened to your talk and I think that only the URB packets are captured, not to isochronous audio ones... but I saw from Google searches that you actually fixed a isochronous packets issue, albeit for IN packets. #21

That particular change caused issues with OUT transfers that were later fixed in https://github.com/desowin/usbpcap/commit/4736a8887743afff3e0bbfe1c920e4db2daa7242. See [mailing list](https://groups.google.com/g/usbpcap/c/SIGXbb1QP08/m/k4RIMGscBwAJ?utm_medium=email&utm_source=footer) for discussion.

> Is there a way to see the OUT audio packets? If not, my only other option is to use a logic analyzer with sigrok and the USB packet decoder, which is much less convenient than Wireshark

If the device is high speed then sigrok won't help (unless the host and/or hub is USB 1.1 only and thus the device operates in full speed).

robcazzaro, 2021-12-21:
First of all, thanks for answering and providing so much information. 

> Is one of the hubs USB2 and the other USB3? If so, with which one does it work and which one it doesn't? Are both hubs bus powered or do they have their own power supply?

It's actually a lot more complicated, with two USB hubs not working (one an old USB 3.0 TekRepublic TUH-300 connected to the USB C port of my T14s Gen 2 Intel laptop, the other a Sabrent HB-TC5P multi type USB 3.0 hub, same port). the latter has a mix of USB 2.0 and 3.0 ports, PD passthru and a HDMI port. Then I have 2 working USB hubs, also connected to the same USB C port on the laptop, one a Inateck USB 3.2 Gen 2 HB2025 (10Gbps), the other a BYEASY USB 3.0

The USB audio DAC is a USB 1.1 full speed device (hence Sigrok might help, even if it's difficult to capture big enough packets that way). I'm enclosing the report from USBTreeView at the end. Outside of the specific ports being connected, USBTreeView sees identical devices independently of the hub working/not working

The only difference I just noticed, is that with the working USB hubs/port, when starting to play music, I see

```
"No.","Time","Source","Destination","Protocol","Length","Info"
"7","7.198828","host","2.22.0","USB","27","Unknown type 7f"
"8","7.199955","2.22.0","host","USB","27","Unknown type 7f"
"9","7.199996","host","2.22.0","USB","39","URB_CONTROL out"
"10","7.200067","2.22.0","host","USB","28","URB_CONTROL out"
"11","7.200508","host","2.22.0","USB","27","Unknown type 7f"
"12","7.200942","2.22.0","host","USB","27","Unknown type 7f"
"13","7.200948","host","2.22.0","USB","39","URB_CONTROL out"
"14","7.201053","2.22.0","host","USB","28","URB_CONTROL out"
"15","7.256033","host","2.22.0","USB","27","Unknown type 7f"
"16","7.257329","2.22.0","host","USB","27","Unknown type 7f"
"17","7.257348","host","2.22.0","USB","39","URB_CONTROL out"
"18","7.257454","2.22.0","host","USB","28","URB_CONTROL out"
"19","7.258498","host","2.22.0","USB","38","URB_CONTROL out"
"20","7.258612","2.22.0","host","USB","28","URB_CONTROL out"
"21","7.258732","host","2.22.0","USB","38","URB_CONTROL out"
"22","7.258856","2.22.0","host","USB","28","URB_CONTROL out"

```

While with one of the non working USB hub, I see a very different flow

```
"No.","Time","Source","Destination","Protocol","Length","Info"
"214","52.241268","2.21.0","host","USB","28","URB_CONTROL out"
"215","61.082907","host","2.21.0","USB","27","Unknown type 7f"
"216","61.083934","2.21.0","host","USB","27","Unknown type 7f"
"217","61.083961","host","2.21.0","USB","39","URB_CONTROL out"
"218","61.084252","2.21.0","host","USB","28","URB_CONTROL out"
"219","61.086706","host","2.21.0","USB","36","SET INTERFACE Request"
"220","61.087516","2.21.0","host","USB","28","SET INTERFACE Response"
"221","62.148719","host","2.21.0","USB","27","Unknown type 7f"
"222","62.149551","2.21.0","host","USB","27","Unknown type 7f"
"223","62.149562","host","2.21.0","USB","39","URB_CONTROL out"
"224","62.149832","2.21.0","host","USB","28","URB_CONTROL out"
"225","62.151229","host","2.21.0","USB","38","URB_CONTROL out"
"226","62.151418","2.21.0","host","USB","28","URB_CONTROL out"
"227","62.151692","host","2.21.0","USB","38","URB_CONTROL out"
"228","62.151868","2.21.0","host","USB","28","URB_CONTROL out"
"229","62.252657","2.21.1","host","USB","27","URB_INTERRUPT in"
"230","62.253868","host","2.21.1","USB","27","URB_FUNCTION_ABORT_PIPE"
"231","62.253912","2.21.1","host","USB","27","URB_INTERRUPT in"
"232","62.253923","2.21.1","host","USB","27","URB_FUNCTION_ABORT_PIPE"
"233","62.382899","host","2.21.1","USB","27","URB_FUNCTION_SYNC_RESET_PIPE_AND_CLEAR_STALL"
"234","62.383493","2.21.1","host","USB","27","URB_FUNCTION_SYNC_RESET_PIPE_AND_CLEAR_STALL"
"235","64.254067","host","2.21.1","USB","27","URB_INTERRUPT in"
"236","64.386735","host","2.21.1","USB","27","URB_INTERRUPT in"
"237","85.376933","host","2.21.0","USB","27","Unknown type 7f"
"238","85.378079","2.21.0","host","USB","27","Unknown type 7f"
"239","85.378104","host","2.21.0","USB","39","URB_CONTROL out"
"240","85.378325","2.21.0","host","USB","28","URB_CONTROL out"
"241","85.380483","host","2.21.0","USB","36","SET INTERFACE Request"
"242","85.381264","2.21.0","host","USB","28","SET INTERFACE Response"
```
But I haven't had time to dig into that yet. At this point it looks as if I'm actually already capturing all I need, and somehow I missed it. I'd appreciate any additional pointer, but is looks as if this was really a PBCAK...

Thanks! Rob

```
[... 67 lines cut ...]
    Device ID            : HID\VID_262A&PID_10B8&MI_00\7&2E70B041&0&0000
    Class                : HIDClass
    Driver KeyName       : {745a17a0-74d3-11d0-b6fe-00a0c90f57da}\0034 (GUID_DEVCLASS_HIDCLASS)
 Child Device 2          : MX3 (USB Audio Device)
  Device Path 1          : \\?\USB#VID_262A&PID_10B8&MI_01#6&27291fee&0&0001#{5ef5d091-6eed-47a3-aa2f-4b3f77520791}\global
  Device Path 2          : \\?\USB#VID_262A&PID_10B8&MI_01#6&27291fee&0&0001#{65e8773d-8f56-11d0-a3b9-00a0c9223196}\global (AM_KSCATEGORY_CAPTURE)
  Device Path 3          : \\?\USB#VID_262A&PID_10B8&MI_01#6&27291fee&0&0001#{65e8773e-8f56-11d0-a3b9-00a0c9223196}\global (AM_KSCATEGORY_RENDER)
  Device Path 4          : \\?\USB#VID_262A&PID_10B8&MI_01#6&27291fee&0&0001#{02baa4b5-33b5-4d97-ae4f-e86dde17536f}\usbaudio1_sideband_ddi_ref
  Kernel Name            : \Device\000001f1
  Device ID              : USB\VID_262A&PID_10B8&MI_01\6&27291FEE&0&0001
  Class                  : MEDIA
[... cut ...]
    Device ID            : SWD\MMDEVAPI\{0.0.0.00000000}.{99CAB632-87FA-403F-A7DA-B10F81F22877}
    Class                : AudioEndpoint
    Driver KeyName       : {c166523c-fe0c-4a94-a586-f1a80cfbbf3e}\0008 (AUDIOENDPOINT_CLASS_UUID)
   Child Device 2        : Line In (MX3) (Audio Endpoint)
    Device ID            : SWD\MMDEVAPI\{0.0.1.00000000}.{ADE88031-CAA7-4134-888F-0720B4F4412C}
    Class                : AudioEndpoint
    Driver KeyName       : {c166523c-fe0c-4a94-a586-f1a80cfbbf3e}\0007 (AUDIOENDPOINT_CLASS_UUID)

```