desowin/usbpcap #121: USBPcap did not recognize URB Function code : Unknown type 7f

jsiobj, 2022-04-18:
Hello,

I am trying to understand how some MIDI controller is working (NI Traktor Kontrol F1) and I am using Wireshark to see USB communications so I can get a teensy board read / write from / to the controller (which is not "pure" MIDI from my understanding). 

And I am also trying to understand how USB and HID work...

At some point I get the following :

```
Frame 86: 27 bytes on wire (216 bits), 27 bytes captured (216 bits) on interface \\.\USBPcap1, id 0
USB URB
    [Source: 1.23.0]
    [Destination: host]
    USBPcap pseudoheader length: 27
    IRP ID: 0xffff9f83619ba570
    IRP USBD_STATUS: USBD_STATUS_SUCCESS (0x00000000)
    URB Function: URB_FUNCTION_GET_MS_FEATURE_DESCRIPTOR (0x002a)
    IRP information: 0x01, Direction: PDO -> FDO
    URB bus id: 1
    Device address: 23
    Endpoint: 0x00, Direction: OUT
    URB transfer type: Unknown (0xff)
        [Expert Info (Error/Malformed): USBPcap did not recognize URB Function code (report to desowin.org/USBPcap)]
            [USBPcap did not recognize URB Function code (report to desowin.org/USBPcap)]
            [Severity level: Error]
            [Group: Malformed]
    Packet Data Length: 0
    [Request in: 85]
    [Time from request: 0.000002000 seconds]
```

I got 4 frames (2 requests / 2 replies) like that :
```
83	6.955347	host	1.23.0	USB	27	Unknown type 7f
0000   1b 00 60 99 89 62 83 9f ff ff 00 00 00 00 2a 00
0010   00 01 00 17 00 00 ff 00 00 00 00

84	6.955368	1.23.0	host	USB	27	Unknown type 7f
0000   1b 00 60 99 89 62 83 9f ff ff 00 00 00 00 2a 00
0010   01 01 00 17 00 00 ff 00 00 00 00

85	6.955376	host	1.23.0	USB	27	Unknown type 7f
0000   1b 00 70 a5 9b 61 83 9f ff ff 00 00 00 00 2a 00
0010   00 01 00 17 00 00 ff 00 00 00 00

86	6.955378	1.23.0	host	USB	27	Unknown type 7f
0000   1b 00 70 a5 9b 61 83 9f ff ff 00 00 00 00 2a 00
0010   01 01 00 17 00 00 ff 00 00 00 00
```

How can I find more info about this "Unknown type". Is it USB or HID related ?

As stated in "Expert Info" in Wireshark, I am reporting this to usb pcap team although I am not sure it is the  right place to 
do it...

Thanks for your help.

Jean-Sébastien.

desowin, 2022-04-18:
> How can I find more info about this "Unknown type". Is it USB or HID related ?

USB related, and more specifically Windows related. This particular one is not really important from overall device functionality point of view, i.e. Linux doesn't even query these descriptors at all. In short, the Microsoft OS descriptors enable USB device to tell what driver should Windows use so Windows can install the driver automatically.

See #78, #115, #121.

jsiobj, 2022-04-18:
Hello @desowin ,
I should have looked a bit more...
Thanks for the quick reply.

closed 2022-04-24 by jsiobj