desowin/usbpcap #128: Windows on ARM64 support

sidd-kishan, 2022-12-21:
Please let us know when we can have an ARM64 version for Windows on ARM OS. As windows on ARM platform is growing it we need tool chain for ARM64 processors to develop the Wi-Fi driver.

michaelwright235, 2023-10-21:
@sidd-kishan I've managed to build this driver for Windows arm64. You can build it yourself with Visual Studio or use my precompiled driver [here](https://github.com/michaelwright235/usbpcap). Works just fine with Wireshark.

tmon-nordic, 2023-10-23:
@michaelwright235 Would you mind to open PR with the Visual Studio project? This is something I wanted to do for long time but never had time to do. The old Windows versions cannot really be supported anymore because appropriate signing is no longer possible.

desowin, 2023-10-23:
@michaelwright235 Would you mind to open PR with the Visual Studio project? This is something I wanted to do for long time but never had time to do. The old Windows versions cannot really be supported anymore because appropriate signing is no longer possible.

benedekkupper, 2024-09-21:
I ran into this as well, and so far I've been unable to use it either with the precompiled driver nor was I able to build it with VS myself.