desowin/usbpcap #131: EXRESOURCE_TIMEOUT_LIVEDUMP

kouzhudong, 2023-06-05:
`0: kd> !analyze -v
*******************************************************************************
*                                                                             *
*                        Bugcheck Analysis                                    *
*                                                                             *
*******************************************************************************

EXRESOURCE_TIMEOUT_LIVEDUMP (1cc)
A kernel ERESOURCE has timed out. This can indicate a deadlock condition or
heavy contention which can cause performance issues.
Arguments:
Arg1: fffff80146e44ba0, The ERESOURCE that has timed out
Arg2: ffffc4079d763040, The thread that detected the timeout
Arg3: 00000000000001f3, The ERESOURCE contention count
Arg4: 0000000000000096, The configured timeout in seconds

Debugging Details:
------------------

Unable to load image \SystemRoot\system32\DRIVERS\USBPcap.sys, Win32 error 0n2
Unable to load image \SystemRoot\System32\drivers\vmusb.sys, Win32 error 0n2

KEY_VALUES_STRING: 1

    Key  : Analysis.CPU.mSec
    Value: 3953

    Key  : Analysis.DebugAnalysisManager
    Value: Create

    Key  : Analysis.Elapsed.mSec
    Value: 24182

    Key  : Analysis.Init.CPU.mSec
    Value: 562

    Key  : Analysis.Init.Elapsed.mSec
    Value: 10630

    Key  : Analysis.Memory.CommitPeak.Mb
    Value: 89

    Key  : WER.OS.Branch
    Value: vb_release

    Key  : WER.OS.Timestamp
    Value: 2019-12-06T14:06:00Z

    Key  : WER.OS.Version
    Value: 10.0.19041.1

FILE_IN_CAB:  ResourceTimeout-20230602-2138.dmp

DUMP_FILE_ATTRIBUTES: 0x10
  Live Generated Dump

BUGCHECK_CODE:  1cc

BUGCHECK_P1: fffff80146e44ba0

BUGCHECK_P2: ffffc4079d763040

BUGCHECK_P3: 1f3

BUGCHECK_P4: 96

FAULTING_THREAD:  ffffc40794dd3040

PROCESS_NAME:  System

[... 194 lines cut ...]
Implicit thread is now ffffc407`94dd3040
  *** Stack trace for last set context - .thread/.cxr resets it
 # RetAddr               : Args to Child                                                           : Call Site
00 fffff801`4641c9c0     : ffff8c80`af200180 ffffc407`94dd3040 00000000`00000000 ffffc407`988c6158 : nt!KiSwapContext+0x76
01 fffff801`4641beef     : ffffc407`94dd3040 00000000`00000000 fffff501`94e14e20 fffff801`4bb0f000 : nt!KiSwapThread+0x500
02 fffff801`4641b793     : fffff801`00000000 00000000`00000000 00000000`00000000 ffffc407`94dd3180 : nt!KiCommitThreadWait+0x14f
03 fffff801`6b692c53     : fffff501`94e14ed0 00000000`00000000 00000000`00000000 11d1c85d`9d7deb00 : nt!KeWaitForSingleObject+0x233
04 fffff801`6b6932c0     : 00000000`00000000 ffffc407`9a348ec0 ffffc407`9a348ea0 00000000`00000000 : USBPcap+0x2c53
05 fffff801`6b693115     : ffffc407`a0b1ea60 fffff501`94e15150 00000000`00000000 ffffc407`9a348ea0 : USBPcap+0x32c0
06 fffff801`46411385     : ffffc407`a0296ca0 fffff501`94e15150 00000000`00000000 ffffc407`00000000 : USBPcap+0x3115
07 fffff801`926d58fd     : ffffc407`a0296ca0 fffff501`94e15150 ffffc407`9a348d50 ffffc407`a0b1ea60 : nt!IofCallDriver+0x55
08 fffff801`46411385     : ffffc407`a0b1ea60 ffffc407`a0296b50 ffffc407`a0987d40 00000000`00000300 : vmusb+0x58fd
09 fffff801`46965b4e     : ffffc407`a0987d40 ffffc407`a167dee0 00000000`00000000 00000000`00000000 : nt!IofCallDriver+0x55
0a fffff801`464045ee     : ffffc407`a0987d40 00000000`00000000 ffffc407`a167dee0 fffff801`4656dd1c : nt!PnpAsynchronousCall+0xea
0b fffff801`4656dba0     : 00000000`00000000 ffffc407`a0987d40 fffff801`4656d250 fffff801`4656d250 : nt!PnpSendIrp+0x9e
0c fffff801`4693b094     : ffffc407`9d841560 00000000`00000000 ffffc407`a167dee0 00000000`00000300 : nt!PnpStartDevice+0x88
0d fffff801`4693b3fb     : ffffc407`9d841560 00000000`00000000 00000000`00000001 fffff801`4656d6f2 : nt!PnpStartDeviceNode+0xec
0e fffff801`46935d69     : ffffc407`9d841560 fffff501`94e15311 00000000`00000000 00000000`00000001 : nt!PipProcessStartPhase1+0x73
0f fffff801`4693f484     : ffffc407`95ce0700 fffff801`46420b01 fffff501`94e153b0 fffff801`00000002 : nt!PipProcessDevNodeTree+0x319
10 fffff801`4656ed56     : 00000001`00000003 ffffc407`8e59ec50 ffffc407`95ce0700 ffffc407`95ce0700 : nt!PiProcessReenumeration+0x88
11 fffff801`4648e5c5     : ffffc407`94dd3040 ffffc407`86093a20 fffff801`46e433c0 00000000`00000000 : nt!PnpDeviceActionWorker+0x206
12 fffff801`465265f5     : ffffc407`94dd3040 00000000`00000080 ffffc407`860c8080 ce08013c`97000000 : nt!ExpWorkerThread+0x105
13 fffff801`466048d8     : ffff8c80`af200180 ffffc407`94dd3040 fffff801`465265a0 bf040000`00ae6703 : nt!PspSystemThreadStartup+0x55
14 00000000`00000000     : fffff501`94e16000 fffff501`94e0f000 00000000`00000000 00000000`00000000 : nt!KiStartSystemThread+0x28
`

kouzhudong, 2023-06-05:
关机卡死的罪魁祸首，发现在这里：C:\Windows\LiveKernelReports，这里有五六个DUMP文件。

C:\Windows\LiveKernelReports>tree /F
卷 Windows 的文件夹 PATH 列表
卷序列号为 86B6-CF2D
C:.
│  ResourceTimeout-20230602-2138.dmp
│
├─ResourceTimeout
│      ResourceTimeout-20230602-2138.dmp
│
├─USBHUB3
│      USBHUB3-20230601-0900.dmp
│      USBHUB3-20230602-2128.dmp
│      USBHUB3-20230605-1017.dmp
│
└─WATCHDOG

C:\Windows\LiveKernelReports>

kouzhudong, 2023-06-05:
而且还经常出现鼠标无法使用的情况。

desowin, 2023-06-05:
Please download https://github.com/desowin/usbpcap/releases/download/1.5.4.0/usbpcap-src-incl-pdb-1.5.4.0.7z and load the USBPcap.pdb file for analysis. After loading pdb, repeat the commands so the offsets turn into actual functions.

kouzhudong, 2023-06-05:
the dump file i was deleted.
in usbpcap\Release\Windows10\x64 no found pdb file.

kouzhudong, 2023-06-05:
USBPcap+0x2c53 is in DkForwardAndWait.

desowin, 2023-06-05:
> the dump file i was deleted. in usbpcap\Release\Windows10\x64 no found pdb file.

Use the file from Windows8. The difference between the two iss the Microsoft signature.