desowin/usbpcap #136: USBPcap - Device address assignment

szoshi, 2024-01-17:
I am testing a custom USB device using USBPCap and Wireshark. I connect three similar devices with the same idVendor and idProduct. Wireshark assigns all three devices the same address, namely 255. The transfers to all three devices are logged from the same two endpoints 2.255.1 and 2.255.2. The devices appear to be working properly. How can I get Wireshark to identify the separate devices with separate addreses? I tried changing the product id on all three devices but that had no effect on the address assignment.

desowin, 2024-01-17:
The problem is not that Wireshark assigns the address 255 but rather that USBPcap doesn't know what address the device had. This can happen when USBPcapDriver doesn't know the endpoint pipe handle. First step in troubleshooting this problem would be to provide means to reproduce it on other systems. This includes at least mentioning the Windows version and exact hardware details.

szoshi, 2024-01-17:
The problem can be reproduced as follows. I am using a Raspberry Pi Pico example called dev_lowlevel. It can be found here:
https://github.com/raspberrypi/pico-examples/blob/master/usb/device/dev_lowlevel/dev_lowlevel.c

I am running it on a Win10 machine. Can this help you to get started? Is there something that is missing in the descriptors perhaps? I can see distinct addresses for the three devices when I find them using PyUSB on the Windows side. USBTreeViewer also shows three distinct devices, as does Windows. The problem only seems to be with USBPcap/Wireshark.

desowin, 2024-01-19:
I am unable to reproduce the issue. See 
[raspberry_pi_pico_dev_lowlevel.zip](https://github.com/desowin/usbpcap/files/13992319/raspberry_pi_pico_dev_lowlevel.zip)
for actual capture I created.

This is with `dev_lowlevel` from `pico-examples` flashed to Raspberry Pi Pico and WinUSB driver installed with Zadig. Then I used pyusb and entered following python code in Python interpreter:
```python
import usb.core
dev = usb.core.find(idVendor=0x0000)
dev.read(0x82, 64)
dev.write(0x01, "Test")
dev.read(0x82, 64)
```

The first read did timeout (correctly reflected in capture with `USBD_STATUS_CANCELED`). The following requests do work as expected.

Therefore the instructions to reproduce are incomplete and/or invalid.

szoshi, 2024-01-20:
Hi, thanks for taking so muxh effort. I can see one point where I am doing things differently. After Assigning my device the WInUSB driver, I noticed that the get_active_configuration call failed. As a result, I was assigning the devices to use libusb. Could that be the reason USBPcap was unable to assign the correct address?

Also, if I have to work with WinUSB, how do I get the active configuration correctly? I am getting a PermissionError. My alternative would be to hardcode the endpoints which is doable since I am developing both sides of the USB bus. Do you foresee any other issues due to this?

desowin, 2024-01-20:
My suspicion would be towards host controller and not the device. Unless you can provide me firmware to flash that does reproduce the problem, I don't quite think it has anything to do with actual device.

desowin, 2024-01-20:
Why would you want to use anything related to configuration? Vast majority of USB devices, if not virtually every one, has only one configuration.and actually WinUSB does not support other than default configuration and to get this working some pretty complex solutions have to be used - see https://github.com/nadavrub/WinUSB-with-non-default-cfg for example.

szoshi, 2024-01-20:
I am clearly no expert at writing PC side software for USB. I am just working with the standard example that came with the pico low level program. I can see in my program now that the configuration is used solely to find the endpoint addresses and since everyone does the USB device and driver at the same time, there should be no need really for this step.

I did try the WinUSB, libusb test on another PC and saw the same results i.e. get_active_configuration does not work with WinUSB but devices are given separate addresses by USBPcap. With libusb, the device addresses are all clubbed together. As of now I can live with hardcoding the endpoints and using WinUSB as this gives me separate traces for multiple devices, which is my goal.

Thank you again for taking the effort to provide a solution.

szoshi, 2024-01-20:
I will close this issue as I have no further questions related to this.

closed 2024-01-20 by szoshi

desowin, 2024-01-21:
> I did try the WinUSB, libusb test on another PC and saw the same results i.e. get_active_configuration does not work with WinUSB but devices are given separate addresses by USBPcap. With libusb, the device addresses are all clubbed together.

What do you mean by "with libusb"? On Windows, libusb can work just fine with WinUSB driver installed. Are you installing some other driver? libusb on Windows works fine for both WinUSB and libusbK drivers that can be installed with Zadig.

szoshi, 2024-01-21:
I used Zadig to assign my device the libusb driver instead of the WinUSB driver. That is when I could read the configuration but all my devices were assigned the address 255.