desowin/usbpcap #137: Device address does not match when system is restored from hibernate

pmyy-wt, 2024-01-24:
When the system is restored from hibernate, USBPcapCMD.exe outputs different device address . But the filtering driver continues to use the old address for filtering.

pmyy-wt, 2024-01-24:
Oh, another problem is the USBPcapCMD.exe also outputs the disabled devices address, which may conflicts with other device.

pmyy-wt, 2024-01-26:
OK, I found message from the tour ([https://desowin.org/usbpcap/tour.html](https://desowin.org/usbpcap/tour.html)) that, before the capture start, the target device needs to disconnect from computer. Maybe it is the reason for the address mismatch after restored from hibernate. But that I came up another problem with the embedded Bluetooth adapter.  I can't unplug it from the computer. I had try to disable and re-enable the Bluetooth adapter, disable/re-enable the USB hub and disable/re-enable the USB controller, all can't capture data from the embedded Bluetooth adaptor.

For the second problem, here a case of conflict address. the disabled device 'Synaptics FP Sensors (WBF) (PID=0017)' has the same address 1 with the device 'USB Composite Device'.
[image]

desowin, 2024-01-26:
> For the second problem, here a case of conflict address. the disabled device 'Synaptics FP Sensors (WBF) (PID=0017)' has the same address 1 with the device 'USB Composite Device'.

This is interesting because this is all enumerated in user-space. Is it possible that it is simply wrong parent-child relationship and the Synaptics FP sensor is really one interface of the composite device?

pmyy-wt, 2024-01-27:
> This is interesting because this is all enumerated in user-space. Is it possible that it is simply wrong parent-child relationship and the Synaptics FP sensor is really one interface of the [...]

As shown in the device manager(image left side), it is an independence device, not a composite device(a laptop embedded fingerprint input device, useless for me, so disabled).

pmyy-wt, 2024-01-27:
I continue research the first problem. The capture always OK when use an external USB Bluetooth dongle. But for the embedded Bluetooth adapter, the capture was only successfully by chance, failure occurred even after reboot. Then I tried passthrough the adapter to vmware client, the capture OK again (on host). Inspect to device manager, the device becomes "VMWare USB Device", and the driver information show usbpcap.sys in it. Compared to its original Bluetooth driver, the usbpcap.sys is not in the driver information. I don't think it is the matter because usbpcap.sys also not in the external dongle's driver information.
[image]
How can I get logs if there some fail in the filter driver?